Nitrokey/nitrokey-3-firmware #18: Improve RGB color spectrum

robin-nitrokey, 2021-11-17:
It looks like the scaling factor for the intensity of the red LED is a bit too low, reducing the red part of RGB colors.  We should try to find better scaling factors.

szszszsz, 2021-11-17:
See below:
- https://github.com/solokeys/solo/pull/80

bordenc, 2022-07-17:
Any chance to make them programmable? So we can configure which events trigger which combinations? That would be keewl.

szszszsz, 2022-07-18:
@bordenc Hi! Sounds interesting! Can you elaborate on that? E.g. what events do you have in mind? Would that be per app, or even more detailed, per app-action type pair?

robin-nitrokey, 2022-07-18:
This would be a nice feature and is definitely feasible.  But realistically, as it would require some effort to design and implement commands for managing the assignments and as it has a low priority compared to OpenPGP/OTP/PWS…, I don’t see this implemented anytime soon (unless you want to do it yourself).

szszszsz, 2022-07-18:
@robin-nitrokey We do have other priorities of course. However the current time constraints do not prevent from taking notes of the idea, while we have it at hand. Implementation specifics and roadmap are orthogonal to it.

robin-nitrokey, 2022-07-18:
Sure, I just wanted to make clear that even if we are talking about details, the chances of having this implemented soon are rather low to avoid false expectations.