Nitrokey/nitrokey-3-firmware #21: Not reacting to the touch button / channel busy

szszszsz, 2021-12-02:
Symptoms from the user POV are: not reacting to the touch button press after a browser monit

Reproduction scenario:
1. Run browser in log taking mode
2. Open https://webauthn.bin.coffee
3. Run both operations

The CCID operations in progress are unknown. PC was rebooted after the first issue occurrence, which has not helped.

Browser log shows that the CTAPHID channel was reported BUSY, thus browser could not communicate with the device.

robin-nitrokey, 2021-12-08:
The only context in which I experienced this problem is if there is a different FIDO2 command that is still waiting for verification (i. e. button press).  If you encounter this issue again, can you check whether the LED is orange or green?

szszszsz, 2021-12-08:
It's green constantly from my experience, perhaps with a different, lighter shade when touching. Note I have development hardware, which could have LED routed improperly.

lukasl93, 2021-12-12:
I can reproduce this behaviour on a freshly delivered Nirtokey 3A shipped from the kickstarter campaign.

When tested with windows 10 and firefox 95.0 the device always shows a green LED and does not react on touch.

szszszsz, 2022-01-03:
Potentially #31 can be a cause.

szszszsz, 2022-01-13:
Unfortunately the cryptolibrary update from #31 has not helped for all the causes. 
Potential cause for the busy issue seems to lie in the implementation on the verge of the trussed and ctaphid handling, which might even result in crash of the device, thus not reacting to any other events later on. 
Current culprits:
- components/usbd-ctaphid/src/pipe.rs:492
- components/usbd-ctaphid/src/pipe.rs:384
- ctaphid-dispatch-29e69aa22a9aea06/6dcbc0d/src/dispatch.rs:63

szszszsz, 2022-01-25:
Potentially fixed by below. To retest.
- https://github.com/solokeys/solo2/commit/ec57c12d9adb2ac0e71ed31699a08badbb8a09bc

commit 0b2f813a (2022-03-16): Merge branch '21-busy-issue' into release-v1.0.2-rc.1

Correct ctaphid keepalive messages

Fixes #21