Nitrokey/nitrokey-3-firmware #34: Make LED signals more distinct

szszszsz, 2022-01-20:
Make LED signals more distinct between operations.
Connected:
- https://docs.nitrokey.com/fido2/windows/index.html#touch-button-and-led-behavior

szszszsz, 2022-06-29:
Continuing from https://github.com/Nitrokey/nitrokey-3-firmware/pull/51:

About LEDs: spec does not say the behavior should be the same.
- https://fidoalliance.org/specs/fido-v2.0-id-20180227/fido-client-to-authenticator-protocol-v2.0-id-20180227.html#usb-hid-wink

> If the authenticator has a display, show the items contained within the user and rp parameter structures to the user. Alternatively, request user interaction in an authenticator-specific way (e.g., flash the LED light). Request permission to create a credential. If the user declines permission, return the CTAP2_ERR_OPERATION_DENIED error.

> 8.1.9.2.1. CTAPHID_WINK (0x08)
The wink command performs a vendor-defined action that provides some visual or audible identification a particular authenticator. A typical implementation will do a short burst of flashes with a LED or something similar. This is useful when more than one device is attached to a computer and there is confusion which device is paired with which connection.

robin-nitrokey, 2022-07-02:
Changed in https://github.com/Nitrokey/nitrokey-3-firmware/pull/51:
- off by default
- white blinking for user confirmation requests
- blue blinking for winking (to be discussed)

robin-nitrokey, 2022-07-06:
Changed in #53:
- single white blink after boot

robin-nitrokey, 2022-07-13:
Changed in #63:
- use white instead of blue for winking (same as user confirmation requests, consistent with Nitrokey FIDO2)

closed 2022-07-13 by robin-nitrokey