Nitrokey/nitrokey-3-firmware #36: fido-authenticator: Google 2FA reported not working

szszszsz, 2022-01-24:
Google's 2FA is reported to not be working (1). Similarly Nextcloud's (2). To investigate.
Firmware: 1.0.1
Browsers: Firefox 96, Chromium 97.0.4692.99

Details:
- https://support.nitrokey.com/t/nitrokey-3c-nfc-and-google-2fa-not-working/3592/7
- https://support.nitrokey.com/t/firmware-update-nitrokey-3/3804/6

szszszsz, 2022-01-24:
Google: can't reproduce it on a freshly flashed Nitrokey 3 1.0.1, development firmware. Logs look correct on communication.
Browser: Chromium, Version 96.0.4664.110 (Official Build) Fedora Project (64-bit), Incognito Window

Perhaps device needs to be reset after update to make this work?

To check:
- [ ] test clean production firmware
- [ ] test upgraded production firmware, path 1.0.0 -> 1.0.1

szszszsz, 2022-01-24:
Nextcloud 22.2.3:
- registered in Firefox, logged in with Chromium Version 97.0.4692.99 (Official Build) snap (64-bit)
- registered in Chromium, and logged in

szszszsz, 2022-01-25:
Working (development) and non-working logs (production) below. Visible difference between the key handles lengths (~1200 vs ~800).
<details>
<summary> Logs </summary>

```
FIDODebug[20:27:16] Received successful U2F sign response from authenticator: 010000001C304502210095DFB3729DE514A0A08B7AF2D2EB9B56B542FBE6ACA09087CD7ED8A83930A79A02202234E005361C39CC27814EDE9C484EDF13F383AC7324F13FF50DD951240E9103

FIDODebug[20:27:13] -> (CTAP2 error code 46)

FIDODebug[20:27:13] <- 2 {1: "google.com", 2: h'DA85B6851D7EF19A749E69ED51E83FBA716D9BB66BDABD5E284A92D992A98800', 3: [{"id": h'44414EBB2A0D2F29FD0D14162269EEA46B644E9B931584A09B4BC833DCC46C7167A1498AFF228C1090132586A59BB8CF', "type": "public-key"}, {"id": h'A3005891D4C293971B2F86F8242E4D1A302E617509501386A2B0B6C008A57EA84AE1B84775198DE73DAFCA55E17AA36CF9F51A71530731F48D254FC507684C19601428289A232A1C2C851D43CEB418AEC3F6A757ADF73D573BA0E9E23F2A267AE3EEEE8B15EB6176E8B67A8F332E9A3C6DFBEB95679A101DF14F1A4E33D5A3637EE4F0418F37D594148A3C362C0CDBCB1ACA012229014C34672B478BD26EA739CE951F025046915EB523267C860780F995F2E8AEF9', "type": "public-key"}], 5: {"up": false}}

FIDODebug[20:27:13] The device supports the CTAP2 protocol.

FIDODebug[20:27:13] -> {1: ["U2F_V2", "FIDO_2_0"], 2: ["credProtect", "hmac-secret"], 3: h'5FCCABBBB3B4B58581DE2695DEF679F4', 4: {"rk": true, "up": true, "credMgmt": true, "clientPin": true}, 5: 7609, 6: [1], 7: 10, 8: 512}

FIDODebug[20:27:12] Sending CTAP2 AuthenticatorGetInfo request to authenticator.

```

```
FIDODebug[20:20:57] Received successful U2F register response from authenticator: 05043BAE5672401189449D4D0F5C2EC9E93FA5409F1C50DE8CAA7C3C07A4135DD5465416C5D3B14CE13156EFE9B7430A62D0ADE0C5C06907EE35F28347BA689DBDFAB5A30058914709EC021E24011FDB9BE9583945581CED9CCD7C56C6E5661A38FA685645CAE32E0F54E8BB566340480EF798700713BAE63993D3B51C227F490EA639DB392C61283A10FDF940A3CC2936BC1C26F1FB75DAC34CB5FBAC9667C78F518C0AA54D407DC36A430E41D11DAED3CFFC1567A2D943A888D60D6737C360A34F970AD1BFF43D3F780B2976A16B5AF09E8E5E640D3574014CED4FD9A645515E85AC88C5EC025024F459F886725B93CC67B7ECCD705D273082038930820171A003020102020304C593300D06092A864886F70D01010B05003036310B300906035504061302444531163014060355040A0C0D4E6974726F6B657920476D6248310F300D06035504030C06526F6F7420333020170D3231313031343135303630365A180F32303731313030323135303630365A3039310B300906035504061302444531163014060355040A0C0D4E6974726F6B657920476D62483112301006035504030C094649444F20434120333059301306072A8648CE3D020106082A8648CE3D030107034200042ABA554082E0E7F68464B233B3BDFEEC3807FFA970B8189C52D09652863B9C262B523DB0D34CA56FAE6F3E76BB230E314AFB85BBAE899059663F9D043D228DA0A3663064301D0603551D0E0416041433815DD1E8655F5F04A6B6EDE0918813DDC219C9301F0603551D23041830168014D3A4D49E69E619A9F4F4A3571572F4E12D50863730120603551D130101FF040830060101FF020100300E0603551D0F0101FF040403020106300D06092A864886F70D01010B050003820201007288682D1DE37715CC17928FEADEF7A849CDCA1F497D70F703C111A56CF259F32ACDFA907CBB19FF4FF4434BA47E52D7AEE442008A92F10A29641293ADE890E2558E3FFFA824A219FDA92FC8A3A90EAE995071EE8EEB510108DBF977FAA4CE3044AECCBFCB18B66439A388CF8E5751D73F012275A083DF84CAE51FD4BA0C9C6963D2DC1AD215E1C726AF9B8602E66302B76BA698E0DF3DA07CB23CED6DEF206035F4D36AE69154F10896781818969865CD339BF245469652AAF49DB47B93DE73FF1ACFF7B9FE097F90EB97B5F2F94FC51549EDE2230059AFEF9D9AF066D034AE3486C646ADCD878E82BF031E1AD6E89818A6E93AD667D07351C47C6AF12B4E2A6970158EDEF0BACCC75953C85E86F0B7A76EEB1AEA8B97BD98C1484B4161CB4753CBDCC7C243FA8ED19AAECE21E95E596524AD7400DB72ABF9AA34C503ED2E319584C9AB29175B2AB1D68BA1FEC2D8A56384C4D1AA4693258097F6B28E54C640CB68B0E4322847AD8C5970F6F9FCB83D6507A20DB183DAA776125BEA4B87B67A9F80EB9B0AC45580310A19F140EC0536BF6E950D70A41A82B785784B297D0665E0200E9CEFEA3164AF4DDCC6ED0ADCE0D66F03412EF2F35E5711897A383C48175D8849CAA30DDB4F5E66D275A1F8B7002F0F855E5A9F86AC994F4ECD20C9E65324D201BAF7C45001C08BC320B68EE83E89E80F1B469C9441D40639377C5CD2463046022100FB1B043C0C79DE88C1A607975220E34F087D5670779E15C9C572E0D72C52041B022100A9A032D100572F6B1C424F6FA18883BB8133069DAFC19F372A1EE6D187C5CBAB

FIDODebug[20:20:55] The device supports the CTAP2 protocol.

FIDODebug[20:20:55] -> {1: ["U2F_V2", "FIDO_2_0"], 2: ["credProtect", "hmac-secret"], 3: h'41414755494430313233343536373839', 4: {"rk": true, "up": true, "credMgmt": true, "clientPin": true}, 5: 7609, 6: [1], 7: 10, 8: 512}

FIDODebug[20:20:55] Sending CTAP2 AuthenticatorGetInfo request to authenticator.

```

</details>

coderkun, 2022-03-13:
Eventhough the release notes for Nitrokey 3 firmware version 1.0.1 say “fido-authenticator: use smaller CredentialID - fixes issues with some services FIDO usage (https://github.com/solokeys/fido-authenticator/pull/8)”, I still get a 500 error on GitLab because the ID has 344 characters (GitLab’s limit is 340 characters).

Is that the same issue as this one?

szszszsz, 2022-03-16:
@coderkun It might be the same, however it works for me at the moment. What is your setup?
- GitLab [14.8.3](https://gitlab.com/gitlab-org/gitlab-foss/-/tags/v14.8.3)
- Nitrokey 3 v1.0.2, production configuration
- Firefox 98

coderkun, 2022-03-16:
GitLab 14.8.3 and Nitrokey 3 v1.0.2 here. I tested Firefoxw 98 and it works there! So it seems other browsers (like Chromium) create a longer CredentialID when registering a new device. However, logging in with an registered device works in Chromium, too.

klumbe, 2022-04-02:
Are there any news about this? It is still not working for Google.
Tried it with various Browsers (chrome-based, Firefox) and OSes (Linux, Windows).
When looking at the Nitrokey forum, it looks like the colleagues are not even aware of the issue anymore.

robin-nitrokey, 2022-04-05:
> Are there any news about this? It is still not working for Google. Tried it with various Browsers (chrome-based, Firefox) and OSes (Linux, Windows). When looking at the Nitrokey forum, it looks like the colleagues are not even aware of the issue anymore.

@klumbe We recently identified the reason for the issue and are currently preparing a firmware update that fixes it.  I can’t announce a release date as of today, but it should be available very soon.

closed 2022-04-07 by robin-nitrokey

robin-nitrokey, 2022-04-07:
The release candidate [1.0.3-rc.1](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.0.3-rc.1) fixes this issue.  We will now run some final tests.  The 1.0.3 release with this fix is scheduled for Monday.

If you want to test the pre-release at your own risk before that, you can download it manually from the Github release page or with `nitropy nk3 fetch-update --version v1.0.3-rc.1` and install it with `nitropy nk3 update irmware-nk3xn-lpc55-v1.0.3-rc.1.sb2`.  Note that you cannot go back to older firmware versions after installing the update.

commit 72155be0 (2022-04-06): Add data migration for FIDO2 attestation cert

This patch adds a data migration step to the runner that updates the
FIDO2 attestation certificate.

Fixes #36.