Nitrokey/nitrokey-3-firmware #40: Hardware repository? Ordering 'hacker' version?

jerabaul29, 2022-03-08:
I cannot find the NK3 hardware repository. Is there one? If not, will it be released open source in the same way as the NK Pro? Also, is it possible to order a 'hacker' version with pins exposed that could be used to develop/ test the firmware?

szszszsz, 2022-03-09:
Hi!
1. Last time I have heard about hardware repository for Nitrokey 3 it was being prepared for release to public. Will check. It seems this was meant for it:
    - https://github.com/Nitrokey/nitrokey-3a-hardware
2. We do not sell hacker version at the moment. However hopefully you should be able to prepare one from the designs once these are published.

For general questions I recommend to use our  [support.nitrokey.com](https://support.nitrokey.com/) where more people can respond, including our community.

Closing as done. Let's continue on forum with related questions.

closed 2022-03-09 by szszszsz

jerabaul29, 2022-03-09:
Thanks :) . Hope the hardware blueprints get released soon :) . Considering a hacker version at some point would be very useful for the community and for helping developers get involved and contribute :) .

I have always disliked a bit forums, as they are often a bit difficult to search, less easy to link to code etc, yet another system to learn, but fair enough ^^ :) .