Nitrokey/nitrokey-3-firmware #44: LED does not go back to normal (green) when using another security token for FIDO2

orolhawion, 2022-04-12:
Steps to reproduce:

1. Plug in a Nitrokey 3 (C NFC in my case)
2. Plug in a different FIDO2 token
3. Log into a FIDO2 secured account and use the different token for FIDO2 authentication
4. Note that the LED of Nitrokey remains the same as if it was waiting for a touch

I tried this with the following different tokens: 
- Yubikey Nano
- Token2 Security Key FIDO2 & U2F

Both different tokens stop blinking when I use the Nitrokey for authentication, so I think the Nitrokey's LED should get back to normal use (green) when the FIDO2 process is over and another token was used.

szszszsz, 2022-04-19:
Connected:
- https://github.com/Nitrokey/nitrokey-3-firmware/issues/34

sosthene-nitrokey, 2023-05-26:
This is a sub-case of #47 and will be fixed by #272. I'm closing it in favor of #47 and #199

closed 2023-05-26 by sosthene-nitrokey