Nitrokey/nitrokey-3-firmware #47: `CTAPHID_CANCEL` doesn't seem to get handled correctly

msirringhaus, 2022-04-27:
This might be related to #44 :

I'm currently implementing CTAP2 into Firefox using [this branch](https://github.com/msirringhaus/authenticator-rs/tree/device_selection_branch). As the branch-name suggests, I just added device selection, when multiple devices are found.
The way this is done (also in Chromium) is to send a 'fake' MakeCredentials-request to every device, making them blink (using pinAuth with zero length, as described in the [CTAP2.0 spec](https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#using-pinToken-in-authenticatorMakeCredential). In CTAP2.1 we got a dedicated command for that.)
Once, one of them gets selected by the user, a `CTAPHID_CANCEL`-command is sent to all other devices, which cancels the pending transaction (and makes the blocking read return). 

Since updating to v1.0.3 (I _think_ it was not there before, but I might be wrong here), when I do this and select some token that is NOT the NK3, my code sends `CTAPHID_CANCEL` to NK3. Then on the next Init, I'm getting `ERR_CHANNEL_BUSY` from NK3.
So NK3 might not handle a cancel correctly?
This code works fine for other tokens (such as YubiKey, but also other vendors).

To test, clone the above branch of the repo and run `cargo run --features crypto_openssl --no-default-features --example ctap2`. In case you want to have more info on what is sent/received, run with `RUST_LOG=trace` (this is very verbose!)

robin-nitrokey, 2022-04-27:
Thanks for the report!  Indeed it seems that `fido-authenticator` is currently not handling the `CTAPHID_CANCEL` command at all:

https://github.com/Nitrokey/nitrokey-3-firmware/blob/8126bc4eaf84bf0e9886eeda28e986be7f8c5de3/components/dispatch-fido/src/fido.rs#L243-245

As this part of the code has been refactored and moved into [fido-authenticator](https://github.com/solokeys/fido-authenticator/blob/main/src/dispatch/ctaphid.rs) recently, we should first merge the upstream changes (#46) before fixing this.

robin-nitrokey, 2024-04-24:
This should be fixed in [v1.7.0-rc.1](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.7.0-rc.1).  Please re-open if it does not work as expected.