Nitrokey/nitrokey-3-firmware #50: Invalid test certificate

szszszsz, 2022-06-28:
While testing against fido2-tests I got multiple errors claiming the attestation certificate is invalid. To check and retest.

- Nitrokey 3 CN, LPC55, v1.0.3-24-ge0b4461, with no-reset-time-window.
- ` fido2.attestation.InvalidData: Attestation certificate must have CA=false!`
- https://github.com/solokeys/fido2-tests/commit/591d3d2279949e08de0766897f24bcfd39af1339

robin-nitrokey, 2022-06-28:
Which certificate did you use?

szszszsz, 2022-06-29:
Works with the updated certificate. Leaving this open to remind about pushing the current development certificates to this repository.

szszszsz, 2023-03-23:
@daringer @robin-nitrokey were the new test certs updated/uploaded? Please close if so.

robin-nitrokey, 2023-03-23:
Not yet.

robin-nitrokey, 2024-04-24:
The certificates are now available at https://github.com/Nitrokey/nitrokey-3-firmware/tree/main/utils/test-certificates.

closed 2024-04-24 by robin-nitrokey