Nitrokey/nitrokey-3-firmware #54: Works poorly on Windows 10

bordenc, 2022-07-03:
This is partly to supplement [this post](https://support.nitrokey.com/t/nitrokey-3-c-2fa-discourse-forum-does-not-register-on-windows-10/4188) and also to respond to the [official documentation](https://docs.nitrokey.com/nitrokey3/windows/passwordless-microsoft.html).

Windows 10 support is very spotty to non-existent. After weeks of following the instructions in the previous links and getting rewarded with "This security key can't be used. Please try a different one," in Windows 10, it (very briefly) started working today when I tried using it in Edge (after failing with Firefox and Chrome). If I unplugged it, plugged it back in, did the right tap sequence and Hermes smiled upon me, Windows would recognise the key. However, I couldn't get it working consistently enough to actually try doing any useful work with it (like logging in to GitHub with it).

The Internet consensus is that Windows support should work, as there are no warnings to say that it's not. Given that the key briefly worked with Edge open, I'm guessing that there is some protocol issue where the communication between the Nitrokey and Windows is out of sync and Windows gives up. However, I don't know where in Event Viewer to start looking for the source of the problem, so I would be grateful for guidance to help troubleshoot.

My system is dual boot and, notwithstanding KDE's limited support for U2F, the exact same hardware works on Linux, so replacing my computer or Nitrokey is currently unnecessary.

szszszsz, 2022-07-04:
Hi! Thank you for the report. 
Indeed on Windows one has to touch the button within the very first second the request comes from the browser, which is not handy to say the least.
We are currently testing v1.0.4 version to address that. Release is scheduled within the next 2 weeks.

szszszsz, 2022-07-04:
Just for completeness / future reference: I have tested a release candidate build of the v1.0.4 today under Windows and I did not see any UX issues.
- https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.0.4-rc.2

bordenc, 2022-07-04:
Curious. Maybe there was a USB reset thingy or something because I definitely got it working (again, one or two times) tapping the key well after 1 second. I assume that this issue/fix also extends to the Windows Hello system, since they all (I'm told) use the same library.

szszszsz, 2022-07-05:
1. Hard to tell. The v1.0.3 problems under Windows are connected with on-device memory usage.
2. The v1.0.4-rc3 beta firmware, signed with the production keys, should be available this week on the proper release page. It can be used with pynitrokey to update by directly specifying the beta version to use.

robin-nitrokey, 2022-07-06:
@bordenc The v1.0.4-rc.3 is now available and you can test it if you want too.  See [this document](https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/testing.md) for more information.

bordenc, 2022-07-06:
Bravo, @robin-nitrokey & team. New firmware is superb. Seems to work perfectly in Windows - and didn't break any of my FIDO2 logins (like on GitHub, for example). I'll let you know if I find anything.

closed 2022-07-06 by bordenc