Nitrokey/nitrokey-3-firmware #65: Unable to login at microsoft.com

jmue, 2022-07-28:
Even with just release firmware version 1.0.4 it is not possible to use the nk3a as a fido2 passwordless authenticator for microsoft.com.
With the newly released version it is possible to register the nk3a at microsoft without any issues (was flawky with prior versions) but it wont be accepted on a subsequent login flow. The site argues that the device is unknown.

Side note: I had no issues to use a Yubikey for the same scenario. Also the nk3a can be used as a fido2 passwordless authenticator for my nextcloud instance and as u2f for numerous sites like google, github or gitlab.

robin-nitrokey, 2022-07-29:
Thanks for the report!  It looks like something changed on Microsoft side because I can reproduce the problem with a stick that previously worked.  Trying to register a new device, I get the error message:

> We encountered an issue setting up your security key.
> We couldn't verify your identity or you are using private mode. Please ensure you are not in a private browsing window and please try again.

Trying to use an already registered device, I get the error message:

> Sign in with a security key
> We couldn't use your device to verify your identity. Enter your password to sign in.

Environment: Chromium 103.0.5060.53 on Debian bullseye

robin-nitrokey, 2022-07-29:
I’ve ran some more tests.  Both a production and a development NK3 worked for me with Google Chrome and Chromium, even in incognito mode.  Only the NK FIDO2 did not work.  @jmue What browser and operating system are you using?  Can you share the exact error message?

Environment:
- Browsers:
  - Google Chrome 103.0.5060.134
  - Chromium 103.0.5060.53
- Devices:
  - NK FIDO2 with firmare v2.3.0
  - Nitrokey 3 production device with firmware v1.0.4
  - Nitrokey 3 development device with firmware g5c1f730

szszszsz, 2022-07-29:
@robin-nitrokey  Just FYI, your Nitrokey FIDO2 firmware is outdated

jmue, 2022-07-29:
I've also ran some tests again. I tested

- chromium 103.0.5060.134-1~deb11u1 on debian 11.4
- edge 103.0.1264.77 on windows 10 21h2
- chromium 103.0.5060.114-1 on fedora 36

I can follow the registration flow as you showed [here](https://docs.nitrokey.com/nitrokey3/windows/passwordless-microsoft.html), bu on a subsequent login the key is not accepted after providing the pin.
[image]

robin-nitrokey, 2022-08-23:
@jmue We suspect there might be an issue if the same device is registered multiple times without deleting the previous resident key.  Please try to execute the following steps:
1. Log in to your Microsoft account and remove your Nitrokey 3 from the list of registered security keys.
2. Open `chrome://settings/securityKeys` in Chromium, select “Sign-in data” and follow the prompts to touch the device and enter your PIN until you see a table with the caption “Security key sign-in data”.  It should contain an entry for the website login.microsoft.com.  Remove it by clicking on the bin icon.
3. Then try again to add the key to your Microsoft account.

jmue, 2022-08-24:
Indeed, you are right. After following your steps, I can log in to microsoft.com with my NK3.

closed 2022-08-24 by jmue