Nitrokey/nitrokey-3-firmware #80: Remove duplicate versions of trussed crate usage

yusdacra, 2022-08-07:
Hello, I'm working on getting this repository work with `dream2nix` (see https://github.com/nix-community/dream2nix/issues/219). I've hit an issue that can't be solved unless either the issue is fixed here or in `cargo` (see https://github.com/nix-community/dream2nix/pull/223#issuecomment-1205500296). Essentially the issue is that duplicate versions of the same dependency (in this case, `trussed`) is not supported by `cargo vendor`, which means `dream2nix` can't build it because it will vendor the dependencies for `cargo` but can't vendor two same versions of a dependency. So I wanted to ask if there is any chance that this duplicate version usage is removed? Or at least one of the sources changed to a different version so this does not happen. I'm testing it on the `runners/lpc55` primarily, which is where I have seen this duplicate versions issue.

robin-nitrokey, 2022-08-08:
Hi @yusdacra!  Good point.  The duplication is caused by [`piv-authenticator`](https://github.com/solokeys/piv-authenticator/blob/1922d6d97ba9ea4800572eea4b8a243ada2bf668/Cargo.toml#L21) still depending on the Trussed Git repository instead of the released v0.1.0 version.  I think we should just remove it for the time being as we don’t use it at the moment anyway.

Out of curiosity, may I ask what your use case for the package is?  Also, we are about to migrate from the `lpc55` runner to the `embedded` runner that works for multiple platforms, see #79.  So it might make sense to work with that one instead.

yusdacra, 2022-08-08:
I don't have any use case myself; I'm just making sure that it builds with `dream2nix` :sweat_smile: Also yeah all of those sound fine to me! I'll be glad to test if the issue is gone if you let me know.

robin-nitrokey, 2022-08-10:
@yusdacra @sbruder The duplication should be fixed with #81.  Let me know if you have any other issues during packaging.

closed 2022-08-10 by robin-nitrokey