Nitrokey/nitrokey-3-firmware #85: Missing firmware-nk3xn-lpc55-v1.2.0.sb2 file release

comio, 2022-09-06:
Dear All,

I'm sorry for this request, maybe something is not clear to me.
I noticed that the 1.2.0 release doesn't provide the sb2 image to use with nitropy.

Thanks in advance for your attention.

Luigi

robin-nitrokey, 2022-09-06:
Hi Luigi,

you are right, the signed update files are not available yet.  We are currently changing our tooling for the creation of these files, therefore it takes a bit longer than usual.  Once they are uploaded, we will change the release type from “Pre-release” to “Release” and update this issue.  Sorry for the confusion!

comio, 2022-09-06:
> Hi Luigi,
>
> you are right, the signed update files are not available yet. We are currently changing our tooling for the creation of these files, therefore it takes a bit longer than usual. Once they [...]

Thanks for the update.

Don't worry, I can wait.

ciao

luigi

fff0x, 2022-09-21:
Is there an ETA, when the latest pre-release will be signed and finally released? Currently it's not possible to leave the Nitrokey 3 (Mini A) attached while working, due to #22. Every action that will access the device using scdaemon is blocked for several minutes. For example when the gpg-agent is involved for SSH connections. This is quite annoying. Many thanks!

zevlee, 2022-09-28:
I noticed that the 1.2.0 version was changed to release. I attempted to update my Nitrokey 3A Mini with nitropy and got the following error:
```
Critical error:
An unhandled exception occurred
	Exception encountered: Exception('Firmware image is not signed by Nitrokey (signed by: unknown)')
```

fff0x, 2022-09-28:
That's due to the pynitrokey checking the signature, just edit `/usr/lib/python3.10/site-packages/pynitrokey/nk3/bootloader/__init__.py`, comment out line 117-120 for now and roll-out the upgrade.

The upgrade itself just works fine. Finally I can leave the dongle attached.

comio, 2022-09-28:
No issues here.

closed 2022-09-28 by comio