Nitrokey/nitrokey-3-firmware #94: NK3AM doesn't enroll with systemd-cryptenroll

runcom, 2022-10-20:
enrolling a NK3ANFC works like a charm:
```
➜  ~ sudo systemd-cryptenroll --fido2-device=auto /dev/nvme0n1p3                            
🔐 Please enter current passphrase for disk /dev/nvme0n1p3: *********************************************************
Initializing FIDO2 credential on security token.
👆 (Hint: This might require confirmation of user presence on security token.)
🔐 Please enter security token PIN: **********              
Generating secret key on FIDO2 security token.
👆 In order to allow secret key generation, please confirm presence on security token.
^C
➜  ~ sudo systemd-cryptenroll --fido2-device=auto /dev/nvme0n1p3
🔐 Please enter current passphrase for disk /dev/nvme0n1p3: *********************************************************
Initializing FIDO2 credential on security token.
👆 (Hint: This might require confirmation of user presence on security token.)
🔐 Please enter security token PIN: **********              
Generating secret key on FIDO2 security token.
👆 In order to allow secret key generation, please confirm presence on security token.
New FIDO2 token enrolled as key slot 1.
```
But if I wipe the slot, and retry with a NK3AM all I get is being stuck and just hangs:
```
➜  ~ sudo systemd-cryptenroll --fido2-device=auto --wipe-slot=fido2 /dev/nvme0n1p3      
FIDO device discovered does not implement FIDO2 with 'hmac-secret' extension.
➜  ~ sudo systemd-cryptenroll --fido2-device=auto --wipe-slot=fido2 /dev/nvme0n1p3
🔐 Please enter current passphrase for disk /dev/nvme0n1p3: *********************************************************
Initializing FIDO2 credential on security token.
👆 (Hint: This might require confirmation of user presence on security token.)
🔐 Please enter security token PIN: **********              
Generating secret key on FIDO2 security token.
👆 In order to allow secret key generation, please confirm presence on security token.
<just hangs here forever w/o any journal or dmesg logs>
```
I've correctly touched the 3AM twice too, and inserted the correct pin too

runcom, 2022-10-20:
So I've heavily patched systemd-crytpenroll to spit out libfido2 debug and this is what I'm getting

<details>
  <summary>More debug</summary>
 
 ```
➜  systemd git:(main) ✗ sudo FIDO_DEBUG=1 SYSTEMD_LOG_LEVEL=debug ./build/systemd-cryptenroll --fido2-device=auto /dev/loop0
FIDO2 device implements extension: credProtect
FIDO2 device implements extension: hmac-secret
FIDO2 device implements option rk: yes
FIDO2 device implements option up: yes
FIDO2 device implements option plat: no
FIDO2 device implements option credMgmt: yes
FIDO2 device implements option clientPin: yes
FIDO2 device implements option credentialMgmtPreview: yes
Has rk ('Resident Key') support: yes
Has clientPin support: yes
Has up ('User Presence') support: yes
Has uv ('User Verification') support: no
Allocating context for crypt device /dev/loop0.
Trying to open and read device /dev/loop0 with direct-io.
Initialising device-mapper backend library.
Trying to load LUKS2 crypt type from device /dev/loop0.
Crypto backend (OpenSSL 3.0.5 5 Jul 2022 [default][legacy]) initialized in cryptsetup library version 2.5.0.
Detected kernel Linux 5.19.15-301.fc37.x86_64 x86_64.
Loading LUKS2 header (repair disabled).
Acquiring read lock for device /dev/loop0.
Opening lock resource file /run/cryptsetup/L_7:0
Verifying lock handle for /dev/loop0.
Device /dev/loop0 READ lock taken.
Trying to read primary LUKS2 header at offset 0x0.
Opening locked device /dev/loop0
Verifying locked device handle (bdev)
LUKS2 header version 2 of size 16384 bytes, checksum sha256.
Checksum:fce76a2f02c13eccf56fc74606fb5b23e1333d252523587c0d866983f00b1a6a (on-disk)
Checksum:fce76a2f02c13eccf56fc74606fb5b23e1333d252523587c0d866983f00b1a6a (in-memory)
Trying to read secondary LUKS2 header at offset 0x4000.
Reusing open ro fd on device /dev/loop0
LUKS2 header version 2 of size 16384 bytes, checksum sha256.
Checksum:2dca645b82d6e792ab94737b5f9535a8f7a7a4660c478924addbe37315619fcc (on-disk)
Checksum:2dca645b82d6e792ab94737b5f9535a8f7a7a4660c478924addbe37315619fcc (in-memory)
Device size 1073741824, offset 16777216.
Device /dev/loop0 READ lock released.
PBKDF argon2id, time_ms 2000 (iterations 0), max_memory_kb 1048576, parallel_threads 4.
Requesting JSON for token 0.
Requesting JSON for token 1.
Requesting JSON for token 2.
Requesting JSON for token 3.
Requesting JSON for token 4.
Requesting JSON for token 5.
Requesting JSON for token 6.
Requesting JSON for token 7.
Requesting JSON for token 8.
Requesting JSON for token 9.
Requesting JSON for token 10.
Requesting JSON for token 11.
Requesting JSON for token 12.
Requesting JSON for token 13.
Requesting JSON for token 14.
Requesting JSON for token 15.
Requesting JSON for token 16.
Requesting JSON for token 17.
Requesting JSON for token 18.
Requesting JSON for token 19.
Requesting JSON for token 20.
Requesting JSON for token 21.
Requesting JSON for token 22.
Requesting JSON for token 23.
Requesting JSON for token 24.
Requesting JSON for token 25.
[... 498 lines cut ...]
here
I'm here yo 0327200512
fido_dev_authkey_tx: dev=0xda2510
fido_tx: dev=0xda2510, cmd=0x10
fido_tx: buf=0xdde540, len=6
0000: 06 a2 01 01 02 02
fido_dev_authkey_rx: dev=0xda2510, authkey=0xddcfb0, ms=7999
fido_rx: dev=0xda2510, cmd=0x10, ms=7999
rx_preamble: buf=0x7fff8539a580, len=64
0000: 00 00 00 03 bf 00 01 04 00 00 00 00 00 00 00 00
0016: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0032: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0048: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
rx_preamble: cid (0x3000000, 0x3000000), cmd (0xbf, 0x10)
rx: rx_preamble
fido_dev_authkey_rx: fido_rx
fido_do_ecdh: fido_dev_authkey
fido_dev_get_assert: fido_do_ecdh
Failed to ask token for assertion: FIDO_ERR_INTERNAL
Releasing crypt device /dev/loop0 context.
Releasing device-mapper backend.
Closing read only fd for /dev/loop0.

  ```
</details>

runcom, 2022-10-20:
```
fido_rx: dev=0xda2510, cmd=0x10, ms=7550
fido_hid_read: fd not ready
rx: rx_preamble
fido_dev_get_assert_rx: fido_rx
```
this is how it ends before trying again in my code

runcom, 2022-10-20:
and that seems to come from https://github.com/Yubico/libfido2/blob/main/src/hid_openbsd.c#L223 - which I clearly don't understand :man_shrugging:

runcom, 2022-10-20:
I really hope the above helps

runcom, 2022-10-21:
Reproducer with just libfido2, notice, if you do the cred + assert operations separately (meaning launching the cred and then assert binaries) the process works. It's when they are in the same process like in the code here that it gets stuck (and this is what happens with systemd basically). Notice also, the 3NFC just works in both scenarios.

```
git clone https://github.com/runcom/libfido2
git checkout nk3m_stuck
# install any build dep for your os
cmake -B build
make -C build
# generate hmac.sercret
head -c 32 /dev/urandom > hmac.secret
# plug the nk
FIDO_DEBUG=1 ./build/examples/nk3m_stuck -P <YOUR_PIN> /dev/hidraw0
# touch twice and it just gets stuck...
# nk3nfc just works
```
if you do the normal split flow with cred + assert then it works (both mini and nfc just works with the commands below):
```
FIDO_DEBUG=1 ./build/examples/cred -k testk.pub -h -P <YOUR_PIN> -v -i cred_id /dev/hidraw0
head -c 32 /dev/urandom > hmac.secret
FIDO_DEBUG=1 ./build/examples/assert -P <YOUR_PIN> -p -h hmac.secret -a cred_id $PWD/testk.pub /dev/hidraw0

runcom, 2022-10-28:
So I got another nk3am and it just works w/o a pin set but when I setup a pin, it doesn't work anymore

robin-nitrokey, 2022-11-02:
Thanks for the report!  I’ll try to reproduce this with my device and get back to you.

runcom, 2022-11-25:
Anything I can help with here? I can 100% reproduce and switching nk between fido2 stuff and gpg stuff is becoming unergonomic

robin-nitrokey, 2022-11-30:
On `6e2ce7ccb8bafd5c8a86b251630dbd74e58d60b2` with default features:

```
panicked at 'no errors: WrongKeyKind', /home/robin/.local/share/cargo/registry/src/github.com-1ecc6299db9ec823/fido-authenticator-0.1.1/src/ctap2.rs:1385:31
```

https://github.com/solokeys/fido-authenticator/blob/0.1.1/src/ctap2.rs#L1385

Returned by: https://github.com/Nitrokey/trussed/blob/v0.1.0-nitrokey-2/src/mechanisms/hmacsha256.rs#L19 because the key is a `P256` key.
Introduced by: https://github.com/Nitrokey/trussed/commit/b0212a4c06fd25310ba5abcf6fc479ebb3abf2ce

Next steps:
1. Check how this is handled on the two-step execution path
2. Think about whether we can work around this in fido-authenticator or whether we have to relax the checks in Trussed.

robin-nitrokey, 2022-11-30:
Okay, so looks like this has nothing to do with the execution path:  In the second example, the hmac-secret extension is just not activated.  If I replace `-h hmac.secret` with `-s hmac.secret`, I get the same panic too.

@sosthene-nitrokey Did you discuss this case when talking about the key kind fixes?

sosthene-nitrokey, 2022-11-30:
What is the key's origin. I think that was what I checked. Do you know what is the stored kind?

robin-nitrokey, 2022-11-30:
It is the credential key, either P256 or Ed255.

runcom, 2022-12-07:
Thanks for this! Looking forward to test it out 🙏🙏🙏

runcom, 2022-12-15:
@robin-nitrokey the latest released firmware still has this issue (30/11) - has the fix landed in the built firmware?

robin-nitrokey, 2022-12-15:
Unfortuantely this has not yet been released.  I’ll let you know once it is.

szszszsz, 2023-02-16:
@robin-nitrokey 
Does this fix the `fido2luks` problems as well? 
Nitrokey 3 can't decode it's hmac-secret request, with serde reporting missing `kty` field (which is a part of COSE I presume).

robin-nitrokey, 2023-02-16:
@szszszsz I don’t think so.  This one only happens with the alpha firmware.  Also, it’s a panic and not a malformed response.  I’ll have to take a closer look at the data fido2luks sends and expects.

szszszsz, 2023-02-16:
Noted. Let's continue this matter on https://github.com/Nitrokey/nitrokey-3-firmware/issues/177 then.

robin-nitrokey, 2023-02-27:
@runcom [v1.2.2-alpha.20230224](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.20230224) contains the fix.

runcom, 2023-02-27:
> @runcom [v1.2.2-alpha.20230224](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.20230224) contains the fix.

awesome, I'll test it out (even if I've been running with the "patch" for quite some time now and it just works :)