Nitrokey/nitrokey-3-firmware #99: OpenPGP support missing

chri2, 2022-10-28:
OpenPGP support missing

The information seems to be more or less available on the forum and in some update blog posts.

But it'd be really nice to have a place to subscribe to the progress or just being informed the moment support for use with gpg is released.

Would be nice if the issue could be confirmed and updated or at least closed when support for OpenPGP / gpg arrives.

szszszsz, 2022-10-28:
Hi!

OpenPGP component for use in the Nitrokey 3 firmware was released publicly lately:
- https://github.com/Nitrokey/opcard-rs/

It is currently in the alpha tests.

Here is the latest blog post about it:
- https://www.nitrokey.com/blog/2022/openpgp-card-alpha-nitrokey-3

runcom, 2022-10-29:
> OpenPGP component for use in the Nitrokey 3 firmware was released publicly lately:

 can you provide build instructions or something to make it clear how the two repositories are linked together?

chri2, 2022-10-29:
Would be nice to have this ticket updated once the support for openpgp is at least public beta and available for all nitrokey3.

szszszsz, 2022-11-02:
@chri2 
I am sure this will be published on blog, as the main communication channel. Can't guarantee the same for this ticket. Let's close it for now to not increase the confusion further.
You can always subscribe to releases RSS, or just add yourself to "Watchers" to follow the development.

@runcom 
As far as I know the OpenPGP support is integrated into Nitrokey 3 Mini already, but I do not know where is the source code for it. The build guide should be the same as the regular one, but simply with the openpgp component added.

runcom, 2022-11-02:
> The build guide should be the same as the regular one, but simply with the openpgp component added.

sure, that's what I'm asking, I want to be able to have docs around how to flash a firmware with opengpg support and test it myself

szszszsz, 2022-11-02:
@daringer Can you share, where the firmware's source code for the beta tests of the OpenPGP component is located? Is it merged to master?

runcom, 2022-11-02:
I'm pretty sure it's here https://github.com/Nitrokey/opcard-rs/ - although the "hard part" is a guide/tutorial/docs on how to build a firmware and flash it to the nk

robin-nitrokey, 2022-11-02:
https://github.com/Nitrokey/opcard-rs is the application, https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.opcard.0.1.0 is the firmware containing the application.  As all sold Nitrokey 3 devices are locked and require signed firmware updates, it is not possible for you to compile and flash your own firmware.  If don’t know if it possible to get unlocked devices at the moment.  Let me know if you would be interested, then I can follow up on this.

Regarding the firmware build instructions:  We currently target two chips, lpc55 and nrf52.  For lpc55, there is a guide: https://github.com/Nitrokey/nitrokey-3-firmware/blob/v1.2.2-alpha.opcard.0.1.0/docs/lpc55-quickstart.md  For nrf52, there is no guide yet but a `Makefile` that takes care of all required steps: https://github.com/Nitrokey/nitrokey-3-firmware/tree/v1.2.2-alpha.opcard.0.1.0/utils/nrf-builder

robin-nitrokey, 2022-11-02:
I’ve created #100 for tracking the opcard integration into the firmware.

closed 2022-11-02 by robin-nitrokey

runcom, 2022-11-02:
> I don’t know if it possible to get unlocked devices at the moment. Let me know if you would be interested, then I can follow up on this.

I'd love to have the possibility to get something for development, since the firmware is open source I believe it would be just inline with the whole philosophy as it allows me to contribute, test and learn