Nitrokey/nitrokey-3-firmware #100: Integrate opcard application

robin-nitrokey, 2022-11-02:
The [opcard](https://github.com/Nitrokey/opcard-rs) application provides an OpenPGP smart card.  This is a tracking issue for its integration into the Nitrokey 3 firmware.  For issues regarding the application itself, see the [opcard issue tracker](https://github.com/Nitrokey/opcard-rs/issues).

Current status:
- 2022-10-13: v0.1.0 alpha release ([app](https://github.com/Nitrokey/opcard-rs/releases/tag/v0.1.0), [firmware](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.opcard.0.1.0), only nrf52)

Roadmap:
- v0.2.0 alpha release: November 2022
- first beta release: tbd
- first stable release: tbd

robin-nitrokey, 2022-12-05:
[v1.2.2-alpha.20221130](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.20221130) released with [ocard v0.2.0](https://github.com/Nitrokey/opcard-rs/blob/main/CHANGELOG.md#020-2022-11-18).

Hoeze, 2023-04-17:
Until when will OpenPGP be implemented?
My lab will soon buy some security tokens and I would prefer if we could get some open-source sticks instead of Yubikeys.

szszszsz, 2023-04-17:
Hey @Hoeze !
We are testing it already in the test firmware, and plan to release it finally in the stable variant in the coming weeks. Target is beginning of May 2023.

Test firmware can be flashed and used by end-user right now, however it is not as much tested as the stable release.
See more about firmware variants at:
- https://github.com/Nitrokey/nitrokey-documentation/pull/146/files?short_path=aa99941#diff-aa999411574c41661c8f3edec54c08e150e804517e4bf010c416c405de676bcc

Hoeze, 2023-04-17:
Thanks @szszszsz, this is great news!
We will evaluate possible options mid May, so this would be just in time :smile:

closed 2023-04-27 by robin-nitrokey