Nitrokey/nitrokey-3-firmware #103: Embedded runner: nrf52840-hal is a git dependency without a `rev` set

sosthene-nitrokey, 2022-11-14:
This causes build issues when running `cargo update`

robin-nitrokey, 2022-11-24:
I think that is fine as we are tracking the upstream main branch.  But it would be good to know why we use a Git dependency in the first place.  Given that is was added 9 months ago and the latest release was 29 days ago, we might be able to use a release instead.

sosthene-nitrokey, 2022-11-24:
> latest release was 29 days ago

And this `nrf52840-hal` release (`0.12`) is incompatible with `nrf52840-pac` version `0.11` which is specified just under and depends on `nrf52840-hal` `0.11`.

robin-nitrokey, 2022-11-24:
Did you check whether we can bump `nrf52840-pac`?

sosthene-nitrokey, 2022-11-24:
I'm not sure, I think I tested and saw that id didn't compile and didn't try further.

PR #115 (2022-11-25): embedded: Use nrf52840-hal 0.15.1
Instead of using nrf52840-hal as a Git dependency, use the published 0.15.1 version.

Fixes #103
commit 51ea2c60 (2022-11-25): embedded: Use nrf52840-hal 0.15.1

Instead of using nrf52840-hal as a Git dependency, use the published
0.15.1 version.

Fixes #103
closed 2022-11-25 by robin-nitrokey