Nitrokey/nitrokey-3-firmware #112: nk3xn alpha bricks lpc55 A and C

freswa, 2022-11-21:
Just flashed the file [alpha-nk3xn-lpc55-v1.2.2-alpha.opcard.0.2.0.oath.0.3.0.sb2](https://github.com/Nitrokey/nitrokey-3-firmware/releases/download/v1.2.2-alpha.opcard.0.2.0.oath.0.3.0/alpha-nk3xn-lpc55-v1.2.2-alpha.opcard.0.2.0.oath.0.3.0.sb2) via `nitropy nk3 update alpha-nk3xn-lpc55-v1.2.2-alpha.opcard.0.2.0.oath.0.3.0.sb2` onto a 3A NFC and a 3C NFC. Update went up to 100% and then a critical error ocurred after a few seconds.
After some sort of (timeout?) the application quited.

Both, the 3A and the 3C aren't even recognized anymore. `dmesg` reports
```
[  +6,892544] usb 4-2: device descriptor read/64, error -110
[ +15,570511] usb 4-2: device descriptor read/64, error -110
[  +0,239442] usb 4-2: new full-speed USB device number 82 using xhci_hcd
[ +15,546919] usb 4-2: device descriptor read/64, error -110
[ +15,573089] usb 4-2: device descriptor read/64, error -110
[  +0,106692] usb usb4-port2: attempt power cycle
[  +0,406841] usb 4-2: new full-speed USB device number 83 using xhci_hcd
[  +0,901844] usb 4-2: Device not responding to setup address.
[  +0,204628] usb 4-2: Device not responding to setup address.
[  +0,207172] usb 4-2: device not accepting address 83, error -71
```
lsusb hangs until I remove both keys.

freswa, 2022-11-21:
Debug Logs:
[3A_nitropy.log](https://github.com/Nitrokey/nitrokey-3-firmware/files/10060884/3A_nitropy.log)
[3C_nitropy.log](https://github.com/Nitrokey/nitrokey-3-firmware/files/10060897/3C_nitropy.log)

daringer, 2022-11-21:
oh, this does not sound good, I've removed the `sb2` release to first investigate what is wrong here, we'll give you further information asap

daringer, 2022-11-22:
Hey @freswa ,
we would like to replace your bricked device(s), please write support (at) nitrokey (dot) com and we'll clarify the details

freswa, 2022-11-22:
As far as I understand these devices still have a working bootloader, but as we access the bootloader via application code and the application is bricked there is no easy way to flash anymore, is there?
Also I could pry them open to solder/attach something or even add a little switch to `Pio0_5` to make the bootloader accessible if that's possible?

freswa, 2022-11-22:
With the help of @daringer I managed to unbrick both devices. Will post a little guide later.

freswa, 2022-11-22:
Personal note, not related to Nitrokey in anyway, ymmv!

So a little guide for anyone who comes here, bricked their NitroKey 3A or 3C and would like to fix it themselves. One note regarding the early 3C cases: It's nearly impossible to reassemble them. You can check your Stick for the version. If the groove on the side of the stick is a waveform you have the early version. If it's straight you're in luck.

1. Pry open the case. A good starting point is left and right of the USB connector. Be careful, don't use excessive force. The two sides are not only connected but there is a tiny (ribbon) cable for the touch button which is glued to the upper side. If that's ripped off you'll be out of luck
2. We have to short the bootloader pin `BL` to `GND` like shown in the two pictures below.
[image]
[image]
4. While shorting `BL`, insert the stick.
5. Flash with nitropy

comio, 2022-11-22:
Why is this bug closed?
Is the issue understood?

Thanks

luigi

freswa, 2022-11-22:
@comio Yes, the firmware files are offline, the problem has been identified, both devices have been recovered and an updated firmware will be released soon hopefully.

comio, 2022-11-23:
Hi @freswa and @daringer,

can you provide any details?
Which was the issue exactly?

ciao

luigi

szszszsz, 2022-11-23:
Hi @comio!
Still working on it. Let me reopen this ticket for now.

daringer, 2022-11-25:
#116 fixes this, to be more precise this one: https://github.com/Nitrokey/nitrokey-3-firmware/commit/eb0ae220fb73123f0626d57ebb1e71ccf1d0c55b

release incoming, currently in internal testing

closed 2022-11-25 by daringer