Nitrokey/nitrokey-3-firmware #119: Dwarf Error: Cannot find DIE

robin-nitrokey, 2022-11-26:
Only occurs with `FEATURES=alpha` on the lpc55, probably caused by a compiler bug related to LTO.  https://github.com/rust-lang/rust/issues/66118 suggests that replacing fat LTO with thin LTO should fix this, but that causes a hardfault for me.  Should be investigated and reported upstream.

> Dwarf Error: Cannot find DIE at 0x53752 referenced from DIE at 0xae6b4

Related tickets:
- https://github.com/rust-lang/rust/issues/41775
- https://github.com/rust-lang/rust/issues/66118

robin-nitrokey, 2023-02-14:
It turns out that using thin LTO actually fixes this.  The hardfault was caused by the stack overflow that we were debugging at the time.  Still, it’s probably better to use fat LTO for firmware releases.  As a short-term solution, we can just set `lto = "thin"` in `Cargo.toml` when debugging.  In the long term, we probably should introduce different build profiles for development and releases.  Maybe we can go back to using the debug profile per default and the release profile in the CI?

https://doc.rust-lang.org/cargo/reference/profiles.html#lto

daringer, 2023-02-14:
Oh there is a ticket for that. @sosthene-nitrokey and I recently also stumbled across this, disabling `lto` fixes it.

Long-term I would also suggest that we disable this only for the `develop` build.
....and in the meantime `lto` should be switched of by-hand for debugging ?

robin-nitrokey, 2023-04-20:
We no longer have to use nightly, so we can close this for the time being.

szszszsz, 2023-07-03:
Moved to:
- https://github.com/rust-lang/rust/issues/66118

closed 2023-09-13 by robin-nitrokey