Nitrokey/nitrokey-3-firmware #122: nk3am 1.2.2 + opcard 0.1.0 suddendly reset

runcom, 2022-11-27:
absolutely nothing happened afaict, certainly I haven't resetted my key and it's now "empty":
```
Reader ...........: Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
Application ID ...: D2760001240103040000A01031250000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: test card
Serial number ....: A0103125
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
UIF setting ......: Sign=off Decrypt=off Auth=off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]
```

runcom, 2022-11-28:
> 1. I can't find this specific firmware in the releases - can you describe the source of it just for the completeness of the report?

I don't understand this :thinking: it's this one https://www.nitrokey.com/blog/2022/openpgp-card-alpha-nitrokey-3

> 2\. We've just released new firmware - could you check if that reproduces to you?

I will

runcom, 2022-11-28:
> I will

actually @szszszsz the release files aren't signed, can I still install that release?

szszszsz, 2022-11-28:
1. Thank you for mentioning - as you see, we have a couple of places with the test firmware right now, and I just wanted to make sure that's the one.
2. That depends on your configuration, but AFAIR testers have switched to signed releases, hence I assume you have as well. Please reach on the support forum to @daringer regarding the details.

runcom, 2022-11-28:
> 2\. That depends on your configuration, but AFAIR testers have switched to signed releases, hence I assume you have as well. Please reach on the support forum to @daringer regarding the details.

I can't install https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.20221125 

```
348        INFO  libusbsio Loading SIO library: /home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/libusbsio/bin/linux_x86_64/libusbsio.so
352        INFO  libusbsio HID enumeration[94181875188000]: initialized
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #0: ThinkPad Thunderbolt 3 Dock USB Audio
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #1: Logitech H820e
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #2: Nitrokey 3
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #3: Kensington Expert Mouse
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #4: Advantage2 Keyboard
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #5: Advantage2 Keyboard
352       DEBUG  libusbsio HID enumeration[94181875188000]: device #6: Advantage2 Keyboard
352        INFO  libusbsio HID enumeration[94181875188000]: finished, total 7 devices
4634      ERROR pynitrokey.nk3.bootloader Failed to parse firmware image
Traceback (most recent call last):
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/__init__.py", line 109, in validate_firmware_image
    metadata = parse_firmware_image(variant, data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/__init__.py", line 132, in parse_firmware_image
    return parse_firmware_image_nrf52(data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/nrf52.py", line 233, in parse_firmware_image
    image = Image.parse(data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/nrf52.py", line 87, in parse
    with pkg.open(Package.MANIFEST_FILENAME) as f:
  File "/usr/lib64/python3.9/zipfile.py", line 1511, in open
    zinfo = self.getinfo(name)
  File "/usr/lib64/python3.9/zipfile.py", line 1438, in getinfo
    raise KeyError(
KeyError: "There is no item named 'manifest.json' in the archive"
4638     WARNING pynitrokey.cli An unhandled exception occurred
Traceback (most recent call last):
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/__init__.py", line 109, in validate_firmware_image
    metadata = parse_firmware_image(variant, data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/__init__.py", line 132, in parse_firmware_image
    return parse_firmware_image_nrf52(data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/nrf52.py", line 233, in parse_firmware_image
    image = Image.parse(data)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/nk3/bootloader/nrf52.py", line 87, in parse
    with pkg.open(Package.MANIFEST_FILENAME) as f:
  File "/usr/lib64/python3.9/zipfile.py", line 1511, in open
    zinfo = self.getinfo(name)
  File "/usr/lib64/python3.9/zipfile.py", line 1438, in getinfo
    raise KeyError(
KeyError: "There is no item named 'manifest.json' in the archive"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/cli/__init__.py", line 111, in main
    nitropy()
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/click/decorators.py", line 38, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/cli/nk3/__init__.py", line 440, in update
    update_version = exec_update(ctx, image, variant)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/cli/nk3/update.py", line 47, in update
    firmware_or_release = _prepare_update(image, current_version, variant)
  File "/home/amurdaca/koding/pynitrokey/venv/lib64/python3.9/site-packages/pynitrokey/cli/nk3/update.py", line 106, in _prepare_update
[... 62 lines cut ...]
4841       INFO  libusbsio HID enumeration[94181875756208]: finished, total 7 devices
[... cut ...]
--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.kncr27x8' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

robin-nitrokey, 2022-11-29:
@runcom Can you please check if `nitropy nk3 test` still works?

runcom, 2022-11-29:
```
➜  ~ /home/amurdaca/koding/pynitrokey/venv/bin/nitropy nk3 test      
Command line tool to interact with Nitrokey devices 0.4.30
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw3

Running tests for Nitrokey 3 at /dev/hidraw3

[1/4]	uuid         	UUID query              	SUCCESS  	3125D4F56E6D0F8E000000000000A1FA
[2/4]	version      	Firmware version query  	SUCCESS  	v1.2.2
[3/4]	provisioner  	Firmware mode           	FAILURE  	Unable to connect with protocol: T0 or T1. Sharing violation.
Please press the touch button on the device ...
[4/4]	fido2        	FIDO2                   	FAILURE  	'x5c'

4 tests, 2 successful, 0 skipped, 2 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.8jwqae6r' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting

```

robin-nitrokey, 2023-02-14:
The cause for this issue has been identified and will be fixed with the next release, see https://github.com/Nitrokey/nitrokey-3-firmware/pull/160.

closed 2023-02-14 by robin-nitrokey