Nitrokey/nitrokey-3-firmware #136: v1.2.2-alpha troubleshooting

bordenc, 2022-12-13:
I'm testing the 1.2.2-alpha series firmware. As advised, I'll put my observations/bugs/etc. into here. I expect most things not to work (and as long as I can revert to stable firmware, I'm happy), so please guide me in providing helpful testing/troubleshooting feedback.

I followed the instructions, including the OpenSC reset, from https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.2.2-alpha.20221130 . Out of the gate, FIDO continues to work (same keys, etc.), but `gpg --card-edit` yields:
```
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```
… and `openpgp-tool --card-info` yields
```
Using reader with a card: Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
AID:             (hexadecimal ID)
Version:         3.4
Manufacturer:    test card
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```

Which is pretty different from this: https://www.nitrokey.com/blog/2022/nitrokey-3-openpgp-card-development

If this is a known issue, I'll await for the next firmware image to retest. Otherwise, I'm happy to provide whatever other troubleshooting information can assist.

robin-nitrokey, 2022-12-13:
@bordenc Thanks for the report!  I don’t think this is a known issue.

@sosthene-nitrokey Can you please have a look at this one?

robin-nitrokey, 2022-12-13:
@bordenc If you want to do some more debugging, you can try [opgpgcard](https://gitlab.com/openpgp-card/openpgp-card/-/tree/main/tools), e. g. `opgpcard status`.  If it also fails, you can enable log messages by setting the `RUST_LOG=debug` environment variable and share them here.

sosthene-nitrokey, 2022-12-13:
Which version of `openpgp-tool` do you have installed? While the fact that `gpg` doesn't recognize the card after a reset probably comes from us, it also looks like we're hitting a bug in `openpgp-tool`, even if it fails it shouldn't cause a buffer overflow like that.

I'd guess both are linked: our card probably sends invalid data that neither `gpg` nor `openpgp-tool` know how to handle.
Can you run `OPENSC_DEBUG=9 openpgp-tool --card-info`? The resulting logs might give more insight as to where the error comes from.

bordenc, 2022-12-13:
`opgpcard status` output:
```
OpenPGP card 0000:#Hex ID# (card version 3.4)

Signature key:
  Algorithm: RSA 2048 [e 32, format 2]
  Signatures made: 0

Decryption key:
  Algorithm: RSA 2048 [e 32, format 2]

Authentication key:
  Algorithm: RSA 2048 [e 32, format 2]

Remaining PIN attempts: User: 0, Admin: 3, Reset Code: 3
```

Seems normal?

All tools are running at the latest Debian Bullseye version:
```
openpgp-tool - OpenPGP card utility version 0.22.0

Copyright (c) 2012-2020 Peter Marschall <peter@adpm.de>
Licensed under LGPL v2
```
```
gpg (GnuPG) 2.2.40
libgcrypt 1.10.1
Copyright (C) 2022 g10 Code GmbH
License GNU GPL-3.0-or-later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /home/user/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2
```

for `OPENSC_DEBUG=9 openpgp-tool --card-info`, I have the output in a 35 kb. Before I share it, is there any identifying information that I should scrub out? I don't speak APDU very well.

bordenc, 2022-12-14:
A little more context: I have an OpenPGP card v. 2.1, which I've been able to use through a USB smart card reader. Therefore, I _should_ have all the necessary Nitrokey prereqs installed and configured. But I haven't used it very often because the libraries aren't great, and I'm not important enough to have much to encrypt, so I forget some commands.

Also, since I'm messing around with this, should I worry about the hardware self-destructing if I enter the wrong PIN too many times? My Firefox browser prompted for the Nitrokey PIN when I started it up, so I entered the PIN I've been using for FIDO… and re-entered it… and re-entered it… and re-entered it… I better not have fried anything!

(I hope that the firmware factory resets and only nukes the keys after X bad PIN attempts)

sosthene-nitrokey, 2022-12-14:
Are you worried about having damaged your Nitrokey or your OpenPGP 2.1 smartcard?

Regarding the Nitorkey: the PINs and the factory reset functionality is independent between applications.
The hardware should not be damaged by the invalid pins.

Given the output of `opgpcard status` It indeed looks like you locked yourself out of the openpgp application and will have to factory reset it again with 

```
opensc-tool -s 00:A4:04:00:06:D2:76:00:01:24:01:00 -s 00:E6:00:00  -s 00:44:00:00
```

This shouldn't impact the FIDO functionality.

It is still intriguing that gpg doesn't detect the card in this state. 

OpenSC version 0.23 has been released, it appears you are using version 0.22, is it possible for you to update?

The logs should not contain any sensitive information besides part of the serial number of the device. You can send only the end of the file where there should not be any identifying information, after the line:

```
AID: d2:76:00:01:24:01:03:04:00:00:a0:20:XX:XX:00:00 
```

Where XX are part of your serial number, the rest should be the same.
The serial number shown by `opgpcard status` should be A020XXXX.

bordenc, 2022-12-14:
> Are you worried about having damaged your Nitrokey or your OpenPGP 2.1 smartcard?

Some security keys and cards (like my OpenPGP card) self-brick if someone tries to brute-force break into them. I just wanted to make sure my Nitrokey wasn't one of those devices.

```
$ opensc-tool -s 00:A4:04:00:06:D2:76:00:01:24:01:00 -s 00:E6:00:00  -s 00:44:00:00

Using reader with a card: Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
Sending: 00 A4 04 00 06 D2 76 00 01 24 01 00 
Received (SW1=0x90, SW2=0x00)
Sending: 00 E6 00 00 
Received (SW1=0x69, SW2=0x85)
Sending: 00 44 00 00 
Received (SW1=0x90, SW2=0x00)
```
(I assume that opensc-tool _does not_ need to be run as root or any special permissions)

Re-running Hello World tests:
```
$ opgpcard status
OpenPGP card 0000:XXXXXXXX (card version 3.4)

Signature key:
  Algorithm: RSA 2048 [e 32, format 2]
  Signatures made: 0

Decryption key:
  Algorithm: RSA 2048 [e 32, format 2]

Authentication key:
  Algorithm: RSA 2048 [e 32, format 2]

Remaining PIN attempts: User: 0, Admin: 3, Reset Code: 3
```
```
$ openpgp-tool --card-info
Using reader with a card: Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
AID:             d2:76:00:01:24:01:03:04:00:00:a0:20:7f:63:00:00
Version:         3.4
Manufacturer:    test card
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```
```
$ gpg --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```
So basically, the exact same output.

> OpenSC version 0.23 has been released, it appears you are using version 0.22, is it possible for you to update?
That, unfortunately, is up to the Debian maintainers. I'm reluctant to tinker with or roll my own packages lest I inadvertently destroy something. They haven't packaged 0.23 in Sid, so I'm hopeful that it won't be subject to the version freeze for Bookworm.

sosthene-nitrokey, 2022-12-14:
Can you try running `gpg --edit-card` with `pcsd` stopped (by stopping the systemd `pcscd.service` and `pcscd.socket`). I've met issues where `gpg` didn't detect hardware keys when pcscd was running.

bordenc, 2022-12-15:
Yep, that got it working:

```
gpg --edit-card 

Reader ...........: 20A0:XXXX:X:0
Application ID ...: D2760001240103040000A0XXXXXX0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: test card
Serial number ....: A020XXXX
Name of cardholder: [not set]
Language prefs ...: [not set]
Salutation .......: 
URL of public key : [not set]
Login data .......: [not set]
Signature PIN ....: forced
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 0 3 3
Signature counter : 0
KDF setting ......: off
Signature key ....: [none]
Encryption key....: [none]
Authentication key: [none]
General key info..: [none]
```

So I'll fiddle some more and report back any errors.

Does this imply that there's a bug in (pcscd)[https://pcsclite.apdu.fr/] that someone should report? Presumably you'll have hundreds of complaints when this firmware goes RC.

sosthene-nitrokey, 2022-12-15:
Great!

The conflict between pcscd and gpg is known behavior. You can use the `disable-ccid` option in your `gpg-agent` config to fix the issue.
See more details: https://dev.gnupg.org/T5632.

The default admin PIN is 12345678 and the default user PIN is 123456. Looking at the `PIN retry counter : 0 3 3` line, you don't have any attempt left for the user PIN. You can reset it with the admin pin which doesn't appear blocked: `gpg --card-edit` then `admin` then `passwd` then `unblock PIN`.

The `openpgp-tool` bug looks like it has already been reported and fixed: https://github.com/OpenSC/OpenSC/issues/2404, so you will need to wait for version 0.23.0 to be available on Debian.

bordenc, 2022-12-15:
Thanks for all of that.

I see that the default PINs are listed at https://docs.nitrokey.com/pro/linux/openpgp-keygen-on-device.html . That info might want to be more obvious so that impatient users (like me) don't miss it and flood you reset requests. The unblocking information should also be obvious, including whether the **Reset Code** is the same as the **Admin PIN** in `passwd`. The (gpg-card documentation)[https://gnupg.org/documentation/manuals/gnupg/gpg_002dcard.html] isn't very helpful at explaining the difference. Is it worth suggesting updates to their documentation, or will they say it's 'vendor specific'? I might attempt a PR on this documentation with my suggestions.

I'll test the `disable-ccid` suggestion. I'm guessing that there's a link between that and the buffer overflow error I got initially, and solving the latter will probably help the former?

Some tests and successes:
- I can set the cardholder name, language pref & salutation. So that seems working properly.
- Attempting to `generate` rsa2048 keys led to a gpg hang. When I pulled the key out, plugged it back in, the keys generated (and I breathed a huge sigh of relief)
- I can change the key attribute to ed25519 for all keys and generate keys. No hang this time, curiously.

I'm happy to play test other features and try to break things if it'll help. At the least, I might find shortcomings in the documentation that need to be fleshed out for new users.

sosthene-nitrokey, 2022-12-15:
Thanks for  the feedback!

If you use `unblock`, gpg will ask you for the resetting code. If you use `passwd` then `unblock` it asks you for the admin PIN, the prompt correctly tells you which one is expected.

> Attempting to generate rsa2048 keys made gpg hang, and the key light stayed solid green. When I pulled the key out, Ctrl+C'd gpg, and plugged it back in, gpg showed valid rsa2048 keys (and I breathed a huge sigh of relief)

That is expected. Since we are currently not using an hardware accelerated RSA implementation, generating an RSA 2048 bit key is particularly slow (but shouldn't take more than 5min). `gpg` does not display any progress indicator (and it can't really). This is also why generating RSA 4096 bit keys is disabled (it would take around 1h). Future releases should be faster.

Elliptic curve cryptography should however be quite fast.

> kdf-setup changed KDF setting from 'off' to 'on', so I assume it worked. In a future project, I'll figure out what [KDF does](https://docs.nitrokey.com/start/linux/setting-kdf-do.html).

KDF-DO is an additional security mechanism that can help protect your PIN against some forms of attacks, by having the host perform some work (a salted password hash) that cannot realistically be done with the limited computing power of the hardware device. Be aware that there are [some issues with it and gpg 2.2](https://dev.gnupg.org/T3891).

We will consider adding some information on how it works in our documentation.

robin-nitrokey, 2023-02-14:
I think all reported issues have been fixed.  Please re-open or create another ticket if I missed something.

closed 2023-02-14 by robin-nitrokey