Nitrokey/nitrokey-3-firmware #145: Feature request: request touching the capacitive 'button' for performing any operation, even if key is pin-unlocked

jerabaul29, 2023-01-03:
Would it be possible, for Nitrokey 3 models that come with a capacitive 'button', to request physically pressing the button for any action to be performed, even if the key has already been pin unlocked / used?

Motivation: for now at work, the policy is that one should unplug their smart card / key as soon as the operation it was needed for has been performed, to avoid that a 'pin unlocked' key perform (unexpected) operations without a (distracted) user (not monitoring the key LED) to be aware. Ie I guess the idea is that, if something is compromised on the computer, even so it will not be able to extract many key-protected passwords, OTP codes, etc, since as soon as the expected one operation to be performed has been done, the key should be unplugged.

There are several annoyances with the policy though: i) that means typing the pin many times a day, which is tiring and repetitive and time consuming ii) that means plugging / unplugging the key many times a day, which has all the annoyances of i), plus in addition a fast wear of the USB connectors on both the computer and key sides.

Hence, the request, since if performing any operation even of a fully unlocked key requires pressing the button, then there is no way a compromised computer can exploit a plugged and unlocked key without the user noticing, which could help change the policy to allowing to keep newer key models plugged in as long as the user is not leaving their computer.

robin-nitrokey, 2023-01-03:
Adding touch verification for each and every operation, even harmless ones like winking, is not planned at the moment.  But I’ll keep your request in mind when we discuss this topic internally.

That being said, we do support requiring a touch button press for important operations.  This details depend on the application:
- Generally, we require a button press for booting into bootloader mode (i. e. applying a firmware update).
- For FIDO2, we require a touch button press for registrations and logins.
- For OpenPGP, there is a UIF flag that you can set to require button presses for sign / decrypt / authenticate operations.
- For OTP and PWS, we plan to make this configurable but this is not implemented or fixed yet.

If you have specific operations that you’d like to have protected, please let me know so we can consider it during the implementation.

jerabaul29, 2023-01-03:
The most important for me would be the important operations you mention be button press protected.

Making sure the default is strong, ie i) the default factory delivered Nk3 requires button press, rather than ii) required button press can be enabled by the user if they want, would be good and solve my need :) .

And making sure it is hard enough to change the behaviour (require admin pin to disable button press requested on important operations?) would be a nice to have, to make sure I / users don't turn this off easily by mistake :) .

jerabaul29, 2023-05-03:
Regarding the UIF: a couple of questions: 1) which default do you plan on applying? (it feels like UIF enabled by default for all 3 would be good if possible), 2) can you mention it somehow in the NK3 documentation / onboarding document when it comes out? :) That would be very useful - for example, I was not aware of this UIF flag.

sosthene-nitrokey, 2023-05-03:
[The specification](https://gnupg.org/ftp/specs/OpenPGP-smart-card-application-3.4.1.pdf) says that disabled should be the default. See § 4.4.1 Data object `D6`:

> User Interaction Flag (UIF) for PSO:CDS (optional):
> If not supported, DO is not available.
> First byte =
> - 00: UIF disabled (**default**)
> - 01: UIF enabled
> - 02: UIF permanently enabled (not changeable with PUT DATA, optional)
> - 03/04: Reserved for caching modes (Yubico)
> Second byte = Content from General feature management ('20' for button/keypad)

sosthene-nitrokey, 2023-05-03:
> can you mention it somehow in the NK3 documentation / onboarding document when it comes out? :) That would be very useful - for example, I was not aware of this UIF flag.

Yes, this will be added to [docs.nitrokey.com](https://docs.nitrokey.com)

jerabaul29, 2023-05-03:
Thanks :) . It feels a bit strange that the default would be to have this off for a key that actually has a capacitive button (is the default disabled a hard rule, or just a reasonable default as not all keys have a button?), but of course you decide :) .

sosthene-nitrokey, 2023-05-03:
It's also for backward compatibility with older host software that would not know that the device expects the user to touch it, which could lead to confusion since  nothing would tell the user to press their device other than the device's LED.

sosthene-nitrokey, 2023-05-26:
- Regarding FIDO: the current behaviour is the one we plan to support.
- Regarding OpenPGP: The documentation on how to setup the UIF flag is available [here](https://docs.nitrokey.com/nitrokey3/linux/openpgp-uif).
- Regarding the secrets app, touch confirmation is currently always prompted.

closed 2023-05-26 by sosthene-nitrokey