Nitrokey/nitrokey-3-firmware #151: usbd-ccid: panicked at 'unexpectedly in receiving state'

sosthene-nitrokey, 2023-01-11:
This bug can be reached with the usbip runner from https://github.com/Nitrokey/nitrokey-3-firmware/pull/149 and running the gnuk test suite

sosthene-nitrokey, 2023-01-13:
It looks like the device is receiving a new command even though it still hasn't fully sent the previous response. It only happens when the `highspeed-usb` feature is not enabled, which makes sense. USB 2.0 packets are always large enough to contain a full APDU.

sosthene-nitrokey, 2023-01-13:
Ok, so it looks like it's actually a simple issue with the gnuk test suite. A test fails, the connection is not cleaned up, and the next test starts running.

sosthene-nitrokey, 2023-01-13:
Ok no, this seems to come from us. When sending an apdu in multiple packets, usbd-ccid sends zlp (zero length packet) even after non-final packets.

If I understand correctly we should send a zlp only after the full apdu has been sent to mark the end of the transaction.

closed 2023-01-23 by sosthene-nitrokey

robin-nitrokey, 2023-02-13:
I still get this when running `gpg --card-status` with the usbip-runner from main.  Does this work for you?

sosthene-nitrokey, 2023-02-14:
I don't have it no.