Nitrokey/nitrokey-3-firmware #171: FIDO implementation incompatible to Apple ID

freswa, 2023-02-09:
Apple launched 2FA via FIDO tokens lately. Adding an NK3 via NFC has some hurdles:

- NFC doesn't work without additional power
- If powered externally via USB, the NFC chip keeps sending a link to the Nitrokey homepage endlessly which triggers a notification for each transmission
- The process of adding the token to the Apple ID returns an "incompatible" token message and refers to the support homepage with a list of supported devices

robin-nitrokey, 2023-02-14:
I’ve previously experienced some issues with NFC when the device is powered externally.  Does this setup work for you with other services?  Does Apple ID work over USB?

freswa, 2023-02-14:
> Does this setup work for you with other services?
For other services I don't even get an error, the keys are just not being recognized.

> Does Apple ID work over USB?
 I'm currently not able to attach a USB device to any of my devices.

robin-nitrokey, 2023-02-14:
Does NFC without external power work with any service?  What kind of device is this?

freswa, 2023-02-14:
> Does NFC without external power work with any service? 

I can't get NFC without external power to work at all, the most I got with 2x 3A and 1x 3C is a operation error, I got once with about 50 tries.

> What kind of device is this?
It's an iPhone 14

robin-nitrokey, 2023-02-14:
Interesting.  Have you already used the Nitrokeys with a different NFC device or only this one?

freswa, 2023-02-14:
Unfortunately that's the only NFC device available. I've used all NKs over USB though.

robin-nitrokey, 2023-02-14:
Okay, thanks for the information.  To summarize:  I’m not surprised that NFC does not work while the device is powered over USB.  I don’t think this is an issue specific to Apple ID, it probably just shows a different error message than other services.

Regular NFC should work though.  Theoretically, the NFC chip on all three Nitrokey could be broken but that would be very unlikely.  There could also be an issue with your iPhone.  I would consider this unlikely too, but just to rule this out:  Do other NFC devices work with our iPhone?  Finally, there could be a compatibility issue between the iPhone 14 and the Nitrokey 3.  I’m not aware of such an issue, but I’ll ask my colleagues for more information.

freswa, 2023-02-14:
I've used two Yubikeys for the Apple ID and some browser services. They work flawlessly with that phone.

jornfranke, 2023-05-23:
I can confirm this problem on iOS 16.5. Unfortunately, the 3C NFC bricked itself afterwards (not sure if this is related to NFC/FIDO2 with IOS or if it simply was its time - it was from the first batch). I used it before mostly via Linux / USB-C. I contacted support already, but they have not come back to me yet.

freswa, 2023-06-12:
NK3 Firmware 1.5.0 works for me. @jornfranke can you confirm?

davegomez, 2023-09-09:
Same issue here with NK3 and firmware 1.5.0

The NFC with my iPhone 14 doesn't work at all while my YubiKey does work flawlessly.

Other services in the NK3 work fine.