Nitrokey/nitrokey-3-firmware #187: NK3 hacker doesn't have opengpg card support anymore with latest code

runcom, 2023-03-04:
Hi, following https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/lpc55-quickstart.md again (reset+build+flash) makes my hacker key working  but opengpg support isn't there anymore:

```
➜  nitrokey-3-firmware git:(main) make -C utils/lpc55-builder flash FEATURES=develop,alpha
...
all good
...
➜  nitrokey-3-firmware git:(main) gpg --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

runcom, 2023-03-06:
one more thing, if I reset and start fresh, when I try to `provision-develop` I get this at the end:
```
...
# Step 2: provision certs
make fw-provision-certs
make[1]: Entering directory '/home/runcom/k/nitrokey-3-firmware/utils/lpc55-builder'
# TODO: add Trussed key & cert
solo2 app provision store-fido-batch-cert data/fido.cert
Error: Empty list of Solo 2
make[1]: *** [Makefile:106: fw-provision-certs] Error 1
make[1]: Leaving directory '/home/runcom/k/nitrokey-3-firmware/utils/lpc55-builder'
make: *** [Makefile:33: provision-develop] Error 2
make: Leaving directory '/home/runcom/k/nitrokey-3-firmware/utils/lpc55-builder'
```

and then listing "fails" like this w/o allowing me to touch the nk:

```
➜  nitrokey-3-firmware git:(main) nitropy nk3 test --exclude provisioner
Command line tool to interact with Nitrokey devices 0.4.33
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/3]	uuid     	UUID query              	SUCCESS  	19DA38B635B58F58BFA26C62DE067D40
[2/3]	version  	Firmware version query  	SUCCESS  	v1.2.2
Please press the touch button on the device ...
[3/3]	fido2    	FIDO2                   	FAILURE  	'x5c'

3 tests, 2 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.hh2x698_' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

robin-nitrokey, 2023-03-06:
Regarding the opcard issue:  GnuPG can be difficult when devices are removed and re-attached, so re-connecting or rebooting the device might already fix it.  Also, please make sure that the alpha feature is really enabled in the build – you should see a `FEATURES:` field in the output of `make flash` that contains `alpha`.

Regarding the provisioning issue:  Does `solo2 ls` show any Nitrokey 3 devices at all?  If not, please try re-installing the latest version.

xundeenergie, 2023-03-20:
gpg is not working. I've 2 nk3 3 with nfs. Both are configured with gpg --edit-card and worked.

After the Upgrade to the last firmware gpg does not recognise them anymore. It's the same on both nk3.

```
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: ccid open error: skip
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: ccid open error: skip
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: ccid open error: skip
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: detected reader 'Alcor Micro AU9540 00 00'
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: detected reader 'Nitrokey Nitrokey 3 [CCID/ICCD Interface] 02 00'
Mär 20 09:13:41 tag-331 gpg-agent[2504195]: scdaemon[2504195]: no supported card application found: Datei oder Verzeichnis nicht gefunden
```

robin-nitrokey, 2023-03-20:
@xundeenergie If you installed the v1.3.0-rc.1 firmware, this is expected as the OpenPGP application is currently only available in alpha.  We will release a new v1.3.0 alpha with OpenPGP support soon.  See https://github.com/Nitrokey/nitrokey-3-firmware/issues/202.  I’m sorry for the confusion.

xundeenergie, 2023-03-20:
2 steps forward, 1 step back...

sorry... i bought my nitrokey 2 years ago... and basic functionality is still not working... :-(

xundeenergie, 2023-03-20:
Is it possible, to flash back to alpha, without losing data on the stick?

runcom, 2023-03-27:
ok, opengpg functionalities are back :+1: althought, I've reprovisioned my nk hacker with:

- reset
- provision-develop
- make -C utils/lpc55-builder flash FEATURES=develop,alpha

and when I run tests with nitropy I get this:

```
➜  nitrokey-3-firmware git:(main) nitropy nk3 test --exclude provisioner                  
Command line tool to interact with Nitrokey devices 0.4.34
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/4]	uuid     	UUID query              	SUCCESS  	19DA38B635B58F58BFA26C62DE06A1FA
[2/4]	version  	Firmware version query  	SUCCESS  	v1.3.0-alpha.20230320
[3/4]	status   	Device status           	SUCCESS  	Status(init_status=<InitStatus.0: 0>, ifs_blocks=81, efs_blocks=478)
Please press the touch button on the device ...
[4/4]	fido2    	FIDO2                   	FAILURE  	'x5c'

4 tests, 3 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.l2mvdslu' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting

➜  nitrokey-3-firmware git:(main) cat /tmp/nitropy.log.l2mvdslu
286        INFO pynitrokey.cli Timestamp: 2023-03-27 12:01:36.414171
286        INFO pynitrokey.cli OS: uname_result(system='Linux', node='fedora', release='6.2.7-300.fc38.x86_64', version='#1 SMP PREEMPT_DYNAMIC Fri Mar 17 16:02:49 UTC 2023', machine='x86_64')
287        INFO pynitrokey.cli Python version: 3.9.16
288        INFO pynitrokey.cli pynitrokey version: 0.4.9
289        INFO pynitrokey.cli cryptography version: 36.0.2
291        INFO pynitrokey.cli ecdsa version: 0.18.0
292        INFO pynitrokey.cli fido2 version: 1.1.0
293        INFO pynitrokey.cli nrfutil version: 6.1.7
293        INFO pynitrokey.cli pyusb version: 1.2.1
294        INFO pynitrokey.cli spsdk version: 1.7.1
322        INFO pynitrokey.cli.nk3.test platform: Linux-6.2.7-300.fc38.x86_64-x86_64-with-glibc2.37
322        INFO pynitrokey.cli.nk3.test uname: uname_result(system='Linux', node='fedora', release='6.2.7-300.fc38.x86_64', version='#1 SMP PREEMPT_DYNAMIC Fri Mar 17 16:02:49 UTC 2023', machine='x86_64')
700        INFO  libusbsio Loading SIO library: /home/runcom/k/pynitrokey/venv/lib64/python3.9/site-packages/libusbsio/bin/linux_x86_64/libusbsio.so
702        INFO  libusbsio HID enumeration[93954025442496]: initialized
702       DEBUG  libusbsio HID enumeration[93954025442496]: device #0: Nitrokey 3
703        INFO  libusbsio HID enumeration[93954025442496]: finished, total 1 devices
827       DEBUG       root print: Found 1 Nitrokey 3 device(s):
827       DEBUG       root print: - Nitrokey 3 at /dev/hidraw0
828       DEBUG       root print: Running tests for Nitrokey 3 at /dev/hidraw0
835       DEBUG       root print: [1/4]	uuid     	UUID query              	SUCCESS  	19DA38B635B58F58BFA26C62DE06A1FA
843       DEBUG       root print: [2/4]	version  	Firmware version query  	SUCCESS  	v1.3.0-alpha.20230320
851        INFO pynitrokey.cli.nk3.test Device status: Status(init_status=<InitStatus.0: 0>, ifs_blocks=81, efs_blocks=478)
852       DEBUG       root print: [3/4]	status   	Device status           	SUCCESS  	Status(init_status=<InitStatus.0: 0>, ifs_blocks=81, efs_blocks=478)
877       DEBUG fido2.server Fido2Server initialized for RP: PublicKeyCredentialRpEntity(name='Example RP', id='example.com')
878       DEBUG fido2.server Starting new registration, existing credentials: 
881       DEBUG       root print: Please press the touch button on the device ...
888       DEBUG fido2.client Register a new credential for RP ID: example.com
910       DEBUG fido2.ctap2.base Calling CTAP2 make_credential
1067      DEBUG  fido2.hid Got keepalive status: 02
1315      DEBUG  fido2.hid Got keepalive status: 02
1563      DEBUG  fido2.hid Got keepalive status: 02
1815      DEBUG  fido2.hid Got keepalive status: 02
[... 10 lines cut ...]
    cert = make_credential_result.attestation_object.att_stmt["x5c"]
KeyError: 'x5c'
3255      DEBUG       root print: [4/4]	fido2    	FIDO2                   	FAILURE  	'x5c'
3255      DEBUG       root print: 4 tests, 3 successful, 0 skipped, 1 failed
3255      DEBUG       root print: Summary: 1 device(s) tested, 0 successful, 1 failed
3256      DEBUG       root print: Critical error:
3256      DEBUG       root print: Test failed for 1 device(s)
3256      DEBUG       root listing all connected devices:
3263      DEBUG       root :: 'Nitrokey FIDO2' keys
3263      DEBUG       root :: 'Nitrokey Start' keys:
3307      DEBUG       root :: 'Nitrokey 3' keys
3309       INFO  libusbsio HID enumeration[93954021458672]: initialized
3309      DEBUG  libusbsio HID enumeration[93954021458672]: device #0: Nitrokey 3
3309       INFO  libusbsio HID enumeration[93954021458672]: finished, total 1 devices
3356      DEBUG       root /dev/hidraw0: Nitrokey 3 19DA38B635B58F58BFA26C62DE06A1FA
3356      DEBUG       root print: --------------------------------------------------------------------------------
[... cut ...]
3356      DEBUG       root print: - Writing an e-mail to support@nitrokey.com is also possible
3356      DEBUG       root print: - Please attach the log: '/tmp/nitropy.log.l2mvdslu' with any support/help request!
3356      DEBUG       root print: - Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

and I'm not sure what is wrong :thinking: as the FIDO2 certificate error is different from https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/lpc55-quickstart.md#flashing-and-provisioning-the-device a

runcom, 2023-03-27:
This seems to be the issue with provision-develop:

```
solo2 app provision store-fido-batch-cert data/fido.cert
Error: Empty list of Solo 2
make[1]: *** [Makefile:139: fw-provision-certs] Error 1
make[1]: Leaving directory '/home/runcom/k/nitrokey-3-firmware/utils/lpc55-builder'
make: *** [Makefile:33: provision-develop] Error 2
make: Leaving directory '/home/runcom/k/nitrokey-3-firmware/utils/lpc55-builder'
```

runcom, 2023-03-27:
so if I run everything manually from the makefile (`$(MAKE) fw-provision-certs`) it works again but really unsure as to why fails as above, might want to add a timeout or something?

runcom, 2023-03-27:
with just a sleep I get the following which is now correct:
```
➜  nitrokey-3-firmware git:(main) ✗ nitropy nk3 test --exclude provisioner       
Command line tool to interact with Nitrokey devices 0.4.34
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/4]	uuid     	UUID query              	SUCCESS  	19DA38B635B58F58BFA26C62DE067D40
[2/4]	version  	Firmware version query  	SUCCESS  	v1.3.0-alpha.20230320
[3/4]	status   	Device status           	SUCCESS  	Status(init_status=<InitStatus.0: 0>, ifs_blocks=74, efs_blocks=478)
Please press the touch button on the device ...
[4/4]	fido2    	FIDO2                   	FAILURE  	Unexpected FIDO2 cert hash for version v1.3.0-alpha.20230320: 71ba065e113f01a1f80035fdf584b9786045db09a0728629ed021b0a8193d1a8

4 tests, 3 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.p61l8n7_' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```
just added a sleep:
```diff
diff --git a/utils/lpc55-builder/Makefile b/utils/lpc55-builder/Makefile
index ab5f470..1f92f9d 100644
--- a/utils/lpc55-builder/Makefile
+++ b/utils/lpc55-builder/Makefile
@@ -29,6 +29,7 @@ provision-develop:
        $(MAKE) bl-flash
        lpc55 reboot
        ./scripts/usbwait.sh 20a0:42b2
+       sleep 10
        # Step 2: provision certs
        $(MAKE) fw-provision-certs
        ./scripts/boot-to-bootrom.sh
```
it seems maybe my laptop is too slow to reload devices and it fails otherwise