Nitrokey/nitrokey-3-firmware #199: NK3Mini: UP is not properly aborted

daringer, 2023-03-15:
It looks like once a User Presence request is started it can only time-out, but if an abort is requested it will not abort. A directly following 2nd UP request will lead to a "busy channel"

To reproduce, e.g. on Gitlab:
1. click "set up new device" 
2. (you see the device asking for UP) 
3.  click "cancel" in the browser 
4. the nk3mini still has the LED active 
5.  click "set up new device" again 
6.  confirm UP 
7.  nothing happens 
8.  "busy channel" (I suspect the device panics) 

aborting works good for the NK3AN (the LED stops and no busy channel)

robin-nitrokey, 2023-03-15:
Interesting that it works with the NK3AN, cf. https://github.com/Nitrokey/nitrokey-3-firmware/issues/47.  Or is this something different?

daringer, 2023-03-15:
I could also imagine that this is originating in the user presence hardware-specific-implementation, which actually makes the most sense if the behavior differs from the nk3an

PR #272 (2023-06-21): Add support for FIDO request cancellation
# Close #199 

## Trussed must accept and check an interrupt flag

- https://github.com/trussed-dev/trussed/pull/125

## Dispatcher and usb drivers needs to exchange the `InterruptFlag` of the running application

Similar work will be needed for the CCID stack too, but is not as urgent, so to simplify we just do CTAPHID  

- https://github.com/trussed-dev/ctaphid-dispatch/pull/4
- https://github.com/trussed-dev/usbd-ctaphid/pull/5

## Backends need adaptation

There is a breaking change. Since the CoreContext holds a reference to the interrupt flag, it [... cut]
closed 2023-06-21 by sosthene-nitrokey