Nitrokey/nitrokey-3-firmware #201: Nitrokey 3A NFC in Windows no longer working correctly with 1.3.0-rc.1

ableeker, 2023-03-16:
I've used Linux to flash the new v1.3.0-rc.1 firmware successfully to my Nitrokey 3A NFC. I've have tested it a bit, and was happy to see it was working. However, when I booted into Windows, it was unfortunately no longer working correctly, even though it did with v1.2.2, and the alpha's.

First I noticed that 'nitropy nk3 list' isn't seeing the key anymore.

Then when I tested the key with sites like webauthn.io, and webauth.me, these sites ask you to verify the registration by pushing the button on the key, but then seem unable to detect that the button has been pressed. Adding the key to Google also failed because it didn't detect the button.

All of this does work in Linux though.

Furthermore, I can't check if it's caused by the firmware, because I can't revert the key to v1.2.2. It refuses to flash with error 'Critical error: The firmware image is older than the firmware on the device.'

robin-nitrokey, 2023-03-17:
Thank you for the report!  We can’t reproduce the problem so far but we will run some more tests.  Do you remember the last alpha version you used on Windows?

ableeker, 2023-03-17:
That will be v1.2.2-alpha.20230224.

I've tried Firefox, and Chrome, but they both didn't work running Windows, while they have done so with earlier firmwares, and still do when running Linux.

Can I factory reset the key? Would `nitropy fido2 reset` do the trick?

sosthene-nitrokey, 2023-03-17:
If `nitropy` can't talk to the device, I doubt resetting the FIDO application will solve the issue.

Can you see the device in the windows [device manager](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c)?

It should show as a USB composite device. You can unplug and replug it to see if it appears/disappears to know it's not some other usb device.

ableeker, 2023-03-17:
It does show up in the Device Manager. It also shows up in the system tray as device 'Nitrokey 3'.

robin-nitrokey, 2023-03-20:
Can you try to execute `nitropy nk3 list` in an admin terminal?

closed 2023-03-20 by ableeker

robin-nitrokey, 2023-03-21:
@ableeker Great to hear that you were able to fix the problem!  Could you share some information on how to check that which drier is used and how you restored it to use the default driver?  Maybe someone else will run into the same problem in the future.

ableeker, 2023-03-21:
When you uninstall incorrect drivers in the Device Manager, they often just reinstall again. These are drivers for Nitrokey 3's, smart cards, ports, and all kinds of USB devices. I've used the following utilities to find and where possible to remove drivers: Device Manager (Show hidden devices), Zadig, USBDeview, Device Cleanup Tool, and Driver Store Explorer. Zadig should show (at least in my case) that Nitrokey 3 (Interface 1) uses driver HidUsb v10, and Nitrokey 3 (Interface 2) uses usbser v10. The other tools can show you which drivers are actually in use, and what their names and locations are. I then used Drive Store Explorer to remove incorrect drivers from the Drive Store, which prevents them from being reinstalled when you try to uninstall them. Now the default drivers should be applied when you uninstall the incorrect ones.

It worked for me!