Nitrokey/nitrokey-3-firmware #202: Nitrokey 3A - OpenGPG functionality won't work after firmware update to 1.3RC1

slymattz, 2023-03-18:
Hi! 

So I've flashed the alpha firmware and that went well. Nitrokey 3A is visible after issuing nitropy nk3 list. I can also ssh into a remote host using an ssh key stored on my NK3A. Yet, gpg --card-status won't talk to my Nitrokey:
`gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
` I've already tried `pkill gpg-agent` and even restarted the machine but to no avail.
Please advise what to do.

sosthene-nitrokey, 2023-03-18:
Hi!

Thank you for the feedback.

This is the expected behaviour.

The alpha firmwares contain experimental applications that are not yet fully ready.
Version 1.3.0-rc.1 on the other hand is a release candidate for the **stable** firmware, and as such contains only application that will be stabilized in the stable release 1.3.0. The OpenPGP application is not yet ready for such stabilization, and is therefore not included in the release candidate, only in the releases marked as alpha contain the current state of OpenPGP functionality.

This release candidate however contains the secrets app (for TOTP), which will be stabilized in 1.3.0.

We apologize for the confusion.

closed 2023-03-18 by sosthene-nitrokey

slymattz, 2023-03-18:
Thank you for your response. IMO, you should've mentioned this important fact either on the blog or here on github. Fortunately, I have a back up solution but there might be other people who've been locked out of their GPG. 

Is there any timeframe for OpenGPG in future releases?

robin-nitrokey, 2023-03-20:
We plan to release a new alpha version with OpenPGP this week.

robin-nitrokey, 2023-03-20:
I’ve also added an item to the release notes explaining this.  We should improve our documentation regarding the difference between stable and alpha releases before the next release.  Sorry for the confusion.

slymattz, 2023-03-21:
@robin-nitrokey Thank you so much! I think I'll wait until the stable firmware with OpenPGP support is released as I've lost my keys anyway. Nice of you, though, to give us an option.

robin-nitrokey, 2023-03-23:
Just to be clear, you should be able to access the keys created with the previous alpha after updating to the new alpha.