Nitrokey/nitrokey-3-firmware #204: Firmware image container

robin-nitrokey, 2023-03-20:
To make it possible to include more metadata with the firmware images, we discussed introducing a custom firmware image container.  I suggest the following format:

- ZIP as a container format for easy compatibility with pynitrokey
- `metadata.json` file with these keys:
  - `version`: the canonical source for version information
  - `pynitrokey` (optional): the minimum pynitrokey version required for this update
  - `images`: a dict mapping variants to filenames inside the container
- `sha256sum` file with the hashes of `metadata.json` and of the firmware images
- potentially a signature (to be discussed)

Example metadata:

```json
{
    "version": "v1.3.0",
    "pynitrokey": "v0.34.0",
    "images": {
        "lpc55": "firmware-nk3xn-lpc55-v1.0.3.sb2",
        "nrf52": "firmware-nk3am-nrf52-v1.0.3.zip"
    }
}
```

daringer, 2023-03-20:
lgtm, some comments:

* should we then also create such a container for all previous releases ?
* I would prefer to use `toml` as this is kind of the upcoming default, right?
* `nitropy nk3 update` would allow overruling the version requirements using `--force` or something similar?

robin-nitrokey, 2023-03-20:
> should we then also create such a container for all previous releases ?

That would allow us to drop support for raw images in pynitrokey and would be much easier to maintain.

> I would prefer to use toml as this is kind of the upcoming default, right?

Generally speaking, I’d prefer it too, but in this case I chose JSON because the file format is pretty simple and reading and writing JSON is supported by all supported Python versions out of the box.  But I don’t have a strong opinion on it.

> nitropy nk3 update would allow overruling the version requirements using --force or something similar?

Probably.  Otherwise we would lock out people who cannot update due to the bledriver dependency.

robin-nitrokey, 2023-03-21:
We should also add a `device` field to the metadata to be able to use the container format with other devices too.

closed 2023-03-31 by robin-nitrokey