Nitrokey/nitrokey-3-firmware #208: nitropy nk3 update to 1.3.0-rc.1 updates but ends with "Critical error"

bordenc, 2023-03-21:
Ran `nitropy nk3 update firmware-nk3xn-lpc55-v1.3.0-rc.1.sb2` and got this output:
```
Command line tool to interact with Nitrokey devices 0.4.25
Current firmware version:  v1.2.2
Updated firmware version:  v1.3.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Perform firmware update: 100%|█...█| 315k/315k [00:04<00:00, 72.2kB/s]

Critical error:
No Nitrokey 3 device found

--------------------------------------------------------------------------------
Critical error occurred, exiting now

Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/user/1000/nitropy.log' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

(Hopefully sanitised) [/tmp/user/1000/nitropy.log](https://github.com/Nitrokey/nitrokey-3-firmware/files/11028055/nitropy.log) attached.

A couple observations about the log:

1. At the beginning, there are various permission errors. Would this be related to [Debian's new enforcement of PEP668](https://peps.python.org/pep-0668/)? If so, there should probably be a link to some guidance on how to configure `pynitrokey` since the Internet isn't very helpful on assisting Debian users with PEP668.
2. At the end of the log (after about 3 megs of data dump) there are multiple attempts to reconnect to the Nitrokey, which all fail. Perhaps the Nitrokey didn't reboot properly?

In a minor panic, I re-ran `nitropy nk3 update firmware-nk3xn-lpc55-v1.3.0-rc.1.sb2` and got this output:
```
Command line tool to interact with Nitrokey devices 0.4.25
Current firmware version:  v1.3.0
Updated firmware version:  v1.3.0
The version of the firmware image is the same as on the device.  Do you want to continue anyway? [y/N]: N
Aborted!
```
So it appears, thankfully, that nothing broke. Also ran `nitropy nk3 test` and got 3/3.

robin-nitrokey, 2023-03-21:
Thanks for the report.  The first boot after the update took a bit longer than usual because of a change to the external flash, so the update command timed out while waiting for the booted device.  We’ve increased the timeout in recent pynitrokey versions and will increase it even more in the next pynitrokey release to avoid this issue.

Your pynitrokey version 0.4.25 is pretty old.  Please make sure to use the latest pynitrokey version when installing firmware updates to avoid compatibility problems.

Regarding the permission errors, this is common behavior and no cause for concern.  pynitroey tries to open all hidraw devices to find Nitrokeys, and some of the devices are not available for a regular user.

robin-nitrokey, 2023-03-21:
Hm, that should install v0.34.0.  Can you share the output of the following commands?

```
$ python3 -V
$ python3 -m platform
$ pip install pynitrokey==
$ pip install pynitrokey== --use-deprecated=legacy-resolver
```

bordenc, 2023-03-21:
```
$ python3 -V
   Python 3.11.2
$ python3 -m platform
   Linux-6.1.0-6-amd64-x86_64-with-glibc2.36
```
(basically Debian testing)
```
$ pip install pynitrokey==
$ pip install pynitrokey== --use-deprecated=legacy-resolver
```
These commands throw the [PEP 668 warning](https://peps.python.org/pep-0668/), as Debian now enforces PEP 668. However, running the venv version of pip3, I got this:
```
ERROR: Could not find a version that satisfies the requirement pynitrokey== (from versions: 0.3.1, 0.3.2, 0.4.0, 0.4.1, 0.4.2, 0.4.3.1, 0.4.4, 0.4.5, 0.4.7, 0.4.8, 0.4.9, 0.4.10, 0.4.11, 0.4.12, 0.4.13, 0.4.15, 0.4.16, 0.4.17, 0.4.18, 0.4.19, 0.4.20, 0.4.21, 0.4.22, 0.4.23, 0.4.24, 0.4.25, 0.4.26, 0.4.27, 0.4.28, 0.4.29, 0.4.30, 0.4.31, 0.4.32, 0.4.33, 0.4.34)
ERROR: No matching distribution found for pynitrokey==
```
on both commands

bordenc, 2023-03-21:
You poor people. You're one of a precious few open source projects that's responsive, courteous and helpful to your users, and you still have to deal with [this nonsense](/NordicSemiconductor/pc-ble-driver-py/issues/230). No worries. I hope the experience report was helpful and, at least, deters others from making the same ones.

Nevertheless, your dedication and professionalism is appreciated and makes me far better about giving you my money :-)

daringer, 2023-03-21:
@bordenc we are trying to fix this nonsense with: https://github.com/Nitrokey/pynitrokey/pull/346 ... this should hopefully enable python 3.11 support

bordenc, 2023-03-23:
> @bordenc we are trying to fix this nonsense with: [Nitrokey/pynitrokey#346](https://github.com/Nitrokey/pynitrokey/pull/346) ... this should hopefully enable python 3.11 support

Can I help at all with the (Windows) testing? I'm happy to mess around with my NK on my own time, but I'm happy to volunteer some testing manpower if I'm directed on what to do.

robin-nitrokey, 2023-03-27:
@bordenc Thank you for that offer!  Can you try to install the suggested patch?  Most interesting would be the behavior with Python 3.11, but Windows would also be useful.

```
$ pipx install --suffix=-git-346 "git+https://github.com/Nitrokey/pynitrokey@refs/pull/346/head#egg=pynitrokey"
$ nitropy-git-346 version
```

Ideally, you could also test `nitropy-git-346 nk3 update` with a NK3AM (even if you just re-install the same version).

Afterwards you can uninstall the patched version with:
```
$ pipx uninstall pynitrokey-git-346
```

bordenc, 2023-03-30:
On Debian Bullseye (fully updated):
```
$ python3 --version
Python 3.11.2

$ nitropy-git-346 version
Command line tool to interact with Nitrokey devices 0.4.34
0.4.34

$ nitropy-git-346 nk3 update alpha-nk3xn-lpc55-v1.3.0-alpha.20230320.sb2
Command line tool to interact with Nitrokey devices 0.4.34
Critical error:
Failed to parse firmware container
        Exception encountered: BadZipFile('File is not a zip file')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/user/1000/nitropy.log.pdy7h3kw' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

Using the LPC55 because I'm on the USB-C.

Windows attempt will be in another post.

robin-nitrokey, 2023-03-30:
@bordenc Thanks for testing!  Sorry, I forgot to mention that.  We also changed the update process so you no longer have to manually download the firmware.  You can just run:

```
$ nitropy-git-346 nk3 update --version v1.3.0-alpha.20230320
```

bordenc, 2023-03-31:
Some day, I'll find a problem that you aren't already two steps ahead on. Revised command works perfectly on Linux. Windows 10 PowerShell output:

```
> python.exe --version
Python 3.9.7

> C:\Users\bordenc\AppData\Roaming\Python\Python39\Scripts\nitropy.exe nk3 update --version v1.3.0-alpha.20230320
Command line tool to interact with Nitrokey devices 0.4.34
Current firmware version:  v1.3.0-alpha.20230320
Updated firmware version:  v1.3.0-alpha.20230320
The version of the firmware image is the same as on the device.  Do you want to continue anyway? [y/N]: y
Do you want to download the firmware version v1.3.0-alpha.20230320? [Y/n]: y
Download v1.3.0-alpha.20230320: 100%|█| 844k/844k [00:01<00:00, 629kB/s]

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Perform firmware update: 100%|█| 426k/426k [00:04<00:00, 85.5kB/s]
Finalize upgrade: 100%|█| 100/100 [00:06<00:00, 15.22%/s]
```

I hope that's useful.

daringer, 2023-04-04:
hey @bordenc , would you like to join our beta-testers? We would be happy to have a Windows user in this group. If you would like to join, please create an account at https://support.nitrokey.com/ and send me a PM there (same nick).

bordenc, 2023-04-05:
@daringer signed up under username `borden` – can't PM you as I'm still on probation, not that I'm complaining.

szszszsz, 2023-04-13:
Looks like you have reached to the solution. Closing as fixed. Please reopen otherwise.