Nitrokey/nitrokey-3-firmware #210: Please certify nitrokey3 at fido Alliance and

hase77, 2023-03-26:
Hi,

I would like to see the Nitrokey3 (and other Fido keys by Nitrokey) certified at the FIDO Alliance (https://fidoalliance.org/certification/) and Microsoft (https://learn.microsoft.com/en-us/azure/active-directory/authentication/concept-fido2-hardware-vendor).

Best regards
Sebastian

pfactum, 2023-04-13:
Why it'd be useful for me personally: in Czech Republic it is possible to access government services online using certified FIDO keys as described here: [Security keys, settings and level of assurance](https://www.mojeid.cz/en/support/statni-sprava/#collapse-security-keys-and-level-of-assurance-43198).

MartinHennrich, 2023-09-06:
> Previous query on this matter: https://support.nitrokey.com/t/fido-authenticator-certification-level-1-l1/4079

This is a similar post: https://support.nitrokey.com/t/nitrokey-fido2-azure-ad/2078

robin-nitrokey, 2025-04-14:
Nitrokey 3 AM is [Level 1 certified](https://www.nitrokey.com/news/2024/nitrokey-3a-mini-receives-official-fido2-certification) by the FIDO Alliance since August 2024 and, with some delay, now also [whitelisted by Microsoft](https://learn.microsoft.com/en-us/entra/identity/authentication/concept-fido2-hardware-vendor#fido2-security-keys-eligible-for-attestation-with-microsoft-entra-id) (make sure to check the English page as the translations are outdated).

closed 2025-04-14 by robin-nitrokey

crepererum, 2025-04-14:
Are there plans to get certifications for the other Nitrokey 3 models, like the Nitrokey 3C NFC?

robin-nitrokey, 2025-04-15:
No specific plans right now, but of course it is something we are considering for the future.