Nitrokey/nitrokey-3-firmware #216: Allow apps to delay/sleep

daringer, 2023-04-02:
That's likely not too easy to accomplish.
* We cannot really pass delay down to the client
* Busy wait using the monotonic timer is also likely not feasible
* Should there be an API call: `WaitFor` ?

sosthene-nitrokey, 2023-04-03:
Having a "sleep" syscall would mean also blocking all other apps from making any form of progress since they run in the same task. Maybe just have a way to get the time, and then the app can keep the time of the latest attempt, compare it to the current time, and if the difference is below a threshold reject the request?

szszszsz, 2023-04-03:
> Maybe just have a way to get the time, and then the app can keep the time of the latest attempt, compare it to the current time, and if the difference is below a threshold reject the request?

1. This would work for my use case.
2. It sounds weird that RTIC does not support yielding. Are you sure?

sosthene-nitrokey, 2023-04-03:
By nature of being a priority based real-time scheduler, RTIC does not have a way to yield and then resume a task outside of starting another higher-priority task, or completing the task and restarting it from the beginning. That's why V2 wants async, so that you can have that type of yielding.

robin-nitrokey, 2023-04-03:
There already is [`ManagementClient::uptime`](https://docs.rs/trussed/latest/trussed/client/trait.ManagementClient.html#method.uptime).

daringer, 2023-04-03:
Ok ok, perfectly fine. 
So then close this issue and anyone who would like to wait should do a busy-wait based on [ManagementClient::uptime](https://docs.rs/trussed/latest/trussed/client/trait.ManagementClient.html#method.uptime) ?

daringer, 2023-04-04:
Closing this as there is nothing to be done inside the firmware, a client/app should use `uptime` to realize time dependent reactions

closed 2023-04-04 by daringer