Nitrokey/nitrokey-3-firmware #217: PC sometimes doesn't see Nitrokey 3A after reboot

proninyaroslav, 2023-04-04:
**OS:** Fedora Linux 37
**Device:** Nitrokey 3A from Fall 2022 batch
**Firmware**: v1.3.0-alpha.20230320

Sometimes if I reboot my PC (not turn on/off, namely reboot), then the system doesn't see the token (when the GDM login window is displayed) and I have to reconnect it so that the system sees it (remove and insert into USB). I'm using udev rule and configured FIDO2 login to the system from this guide: https://docs.nitrokey.com/fido2/linux/desktop-login

szszszsz, 2023-04-04:
Hi!
Does the red LED shine on the device, when it is not detected?

proninyaroslav, 2023-04-04:
No, it doesn't give any signs.

proninyaroslav, 2023-04-04:
Now I tried to reboot and got the same thing. Pynitrokey does not see it, it's not in dmesg. Only extracting the key helps.

daringer, 2023-04-04:
This sounds pretty much like a hardware issue, please write to support (at) nitrokey (dot) com with your sales order ID and a reference to this issue and we'll make sure that you get a fully working device.

closed 2023-04-04 by daringer

szszszsz, 2023-04-05:
@proninyaroslav Just a quick question or two: 
1. Did it behave like that from the beginning?
2. Have you made any firmware update in the meantime?
3. Can you confirm that the model you mean is Nitrokey 3A NFC (and not Nitrokey 3A Mini)?

proninyaroslav, 2023-04-05:
1. Probably yes, I started using the key a couple of weeks ago, before that it just lay in the locker idle.
2. I immediately updated it to the current alpha version, since I need OpenPGP support. So I couldn't test it on stable.
3. Yes, Nitrokey 3A NFC.

szszszsz, 2023-04-05:
Thank you for the quick update. Please follow the mentioned instructions to get the device replaced.