Nitrokey/nitrokey-3-firmware #227: Misssing OpenPGP support in v1.3.1?

comio, 2023-04-07:
Hi Nitrokey Team,

I just upgraded to v1.3.1 firmware my NK3 Nano. The gpg/openpgp card support stopped to work.

```
gpg --card-status 
gpg: selecting card failed: Nessun device corrisponde
gpg: Scheda OpenPGP non disponibile: Nessun device corrisponde

```
I tried to reset via opensc-tool also.

Is the OpenPGP Card support missing in v1.3.1?

Thanks a lot.

ciao

luigi

Chimrod, 2023-04-07:
Hello I have the same issue with a batch comming from the pre-order:

```
$ nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.36
UUID:               B9EF61DC738B0E569C389B06AA249067
Firmware version:   v1.3.1
Init status:        ok
Free blocks (int):  60
Free blocks (ext):  478

$ gpg --card-status 
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

sosthene-nitrokey, 2023-04-07:
Hi,

OpenPGP support is not stable yet, and is therefore only available in the alpha firmware releases. We will soon release a 1.3.1 alpha version so that you can get OpenPGP functionality back.

comio, 2023-04-07:
@sosthene-nitrokey 

I noticed the missing support. Can you upgrade the release page with a banner with this missing?
Thanks
luigi

robin-nitrokey, 2023-04-07:
I’ve added it to the known issues.

closed 2023-04-07 by comio

robin-nitrokey, 2023-04-18:
@comio [v1.3.1-test.20230417](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.3.1-test.20230417) with OpenPGP support is now available.