Nitrokey/nitrokey-3-firmware #229: Secrets app giving other OTPs than password manager with the same secret

orolhawion, 2023-04-11:
I have issues with getting the secrets app to work on my Nitrokey 3C NFC, or at least it is not working as I would expect it to. I did the following:

> $ nitropy nk3 version
Command line tool to interact with Nitrokey devices 0.4.36
v1.3.1

> $ nitropy nk3 secrets register MyApp $(base32 MyApp)
Command line tool to interact with Nitrokey devices 0.4.36
Please touch the device if it blinks
Current Password (8 attempts left): 

> $ nitropy nk3 secrets list         
Command line tool to interact with Nitrokey devices 0.4.36
Please touch the device if it blinks
Current Password (8 attempts left): 
b'MyApp'

> $ nitropy nk3 secrets get MyApp                     
Command line tool to interact with Nitrokey devices 0.4.36
Please touch the device if it blinks
Current Password (8 attempts left): 
Timestamp: 2023-04-11T08:45:56 (1681195556), period: 30
204269

The file 'MyApp' contains the trimmed OTP secret.

The OTP 204269 is different from what my password manager shows and is invalid for login. 

What am I doing wrong?

robin-nitrokey, 2023-04-11:
The service might be using a different hash algorithm.  You can use the `--hash` option for `register` to select a different algorithm (default: `SHA1`).  If this does not help, we would need to know the service you use.  If you don’t want to share it here, you can also send a mail to me directly or to support@nitrokey.com.

orolhawion, 2023-04-12:
I tried Gitlab and GitHub, both are wrong. I also added the secret to my YubiKey 5 with the default options (sha1, 6 digits, totp), the codes from the YubiKey are different from the Nitrokey codes and valid for login. can you reproduce this with a GitHub account?

robin-nitrokey, 2023-04-12:
I just tested it on GitHub and I did not have any issues.

> the only difference between the Nitrokey and YubiKey procedure is that for Nitrokey I have to provide the secret base32 encoded whereas the YubiKey just takes the secret string.

How do you determine the secret?  The secret in the QR code generated by GitHub already is Base32-encoded.  Maybe you just have to drop the encoding step?

orolhawion, 2023-04-12:
skipping the base32 part fixed it for GitHub but not for Gitlab.

szszszsz, 2023-04-12:
Hi!
The OTP implementation used in the Nitrokey 3 firmware was tested against the test vectors provided in the specification for SHA1, so I expect this is some pynitrokey's UI/UX issue.
Can you provide the rest of the URL (with the secret removed) for Gitlab? Perhaps other parameters have changed as well, like the mentioned algorithm or the number of digits.

I think it would be a good idea to have the whole URL interpreted by pynitrokey directly, instead of asking user to divide it properly.
Moving to pynitrokey project.