Nitrokey/nitrokey-3-firmware #235: FIDO2 tests fail for v1.3.1-test.20230414

robin-nitrokey, 2023-04-16:
With v1.3.1-test.20230414, calling GetAssertion for a credential that was created using the same firmware version causes a CTAP error 0x7F (OTHER).

Reproduction:
- `python3 test-release.py v1.3.1-test.20230414` with https://github.com/nitrokey/nitrokey-3-tests using the usbip runner
- `nitropy nk3 test --only fido2` with hardware

Logs:
<details>
<summary>Backtrace from nitrokey-3-tests</summary>

```
________________________________ TestFido2.test ________________________________

self = <fido2.client.Fido2Client object at 0x7f7451d770d0>
options = PublicKeyCredentialRequestOptions(challenge=b'jlL\xe6\xd41O\xebO\xe4u<\xcam\xaao\xe7\x1b\x17\xb6\xb3\xf9\xd8\xc9&;\xad...(\x90', transports=None)], user_verification=<UserVerificationRequirement.DISCOURAGED: 'discouraged'>, extensions=None)
event = <threading.Event object at 0x7f7451d89eb0>

    def get_assertion(
        self,
        options: PublicKeyCredentialRequestOptions,
        event: Optional[Event] = None,
    ) -> AssertionSelection:
        """Get an assertion.
    
        :param options: PublicKeyCredentialRequestOptions data.
        :param threading.Event event: (optional) Signal to abort the operation.
        """
    
        options = PublicKeyCredentialRequestOptions.from_dict(options)
        event = event or Event()
        if options.timeout:
            timer = Timer(options.timeout / 1000, event.set)
            timer.daemon = True
            timer.start()
    
        logger.debug(f"Assert a credential for RP ID: {options.rp_id}")
        self._verify_rp_id(options.rp_id)
    
        client_data = self._build_client_data(
            CollectedClientData.TYPE.GET, options.challenge
        )
    
        try:
>           return self._backend.do_get_assertion(
                client_data,
                options.rp_id,
                options.allow_credentials,
                options.extensions,
                options.user_verification,
                event,
            )

/venv/lib/python3.9/site-packages/fido2/client.py:836: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <fido2.client._Ctap2ClientBackend object at 0x7f7451d77160>
client_data = CollectedClientData(type='webauthn.get', challenge=b'jlL\xe6\xd41O\xebO\xe4u<\xcam\xaao\xe7\x1b\x17\xb6\xb3\xf9\xd8\xc9&;\xad\xd9\x16\xdb+\xdf', origin='https://example.com', cross_origin=False)
rp_id = 'example.com'
allow_list = [PublicKeyCredentialDescriptor(type=<PublicKeyCredentialType.PUBLIC_KEY: 'public-key'>, id=b'\xa3\x00X\x98\xaaP&\xda#\...f4\xae\xf9\x01L\xc3c\xb1\xad\x8b<\x04\xc0\xe8\xc1#\xf0\x02P\xb8\x0bL\x7f6\xd8\xfe\xbb;hi|\xa3z(\x90', transports=None)]
extensions = None
user_verification = <UserVerificationRequirement.DISCOURAGED: 'discouraged'>
event = <threading.Event object at 0x7f7451d89eb0>

    def do_get_assertion(
        self,
        client_data,
        rp_id,
        allow_list,
        extensions,
        user_verification,
[... 336 lines cut ...]
 INFO  usbip_runner                   > >>>> Received confirmation request. Confirming automatically.
 INFO  usbip_runner                   > Set status: Idle
 INFO  usbip_runner                   > Set status: Idle
 INFO  usbip_runner                   > Set status: Processing
 INFO  usbip_runner                   > Set status: Idle
 INFO  !                              > key result
 INFO  usbd_ctaphid::pipe             > Got 1 bytes response from authenticator, starting send
```
</details>

<details>
<summary>Client logs from nitrokey-3-tests</summary>

```
DEBUG    fido2.server:server.py:153 Fido2Server initialized for RP: PublicKeyCredentialRpEntity(name='Example RP', id='example.com')
DEBUG    fido2.server:server.py:187 Starting new registration, existing credentials: 
DEBUG    fido2.client:client.py:772 Register a new credential for RP ID: example.com
DEBUG    fido2.ctap2.base:base.py:361 Calling CTAP2 make_credential
DEBUG    fido2.server:server.py:290 Verifying attestation of type packed
INFO     fido2.server:server.py:297 New credential registered: a3005898aa5026da23889da587cd33b2f3345270557326d2cfd41c836f50057ba636bb2693401d5a8d78ce5f023b4c28e2b3c2c074f4a622264904ee566d25034e600b44167bbd18e993a24d9483045bc1ee5599f45badd37573527de825ba0715cbcd61378de7ac4e947bde02ab813f312eecc15fab2d13c44fb8c654e879873474b762cf402d0eca7ee8a6adfe98edf6e1a3e0b7b4890f46f4aef9014cc363b1ad8b3c04c0e8c123f00250b80b4c7f36d8febb3b68697ca37a2890
DEBUG    fido2.server:server.py:329 Starting new authentication, for credentials: a3005898aa5026da23889da587cd33b2f3345270557326d2cfd41c836f50057ba636bb2693401d5a8d78ce5f023b4c28e2b3c2c074f4a622264904ee566d25034e600b44167bbd18e993a24d9483045bc1ee5599f45badd37573527de825ba0715cbcd61378de7ac4e947bde02ab813f312eecc15fab2d13c44fb8c654e879873474b762cf402d0eca7ee8a6adfe98edf6e1a3e0b7b4890f46f4aef9014cc363b1ad8b3c04c0e8c123f00250b80b4c7f36d8febb3b68697ca37a2890
DEBUG    fido2.client:client.py:828 Assert a credential for RP ID: example.com
DEBUG    fido2.ctap2.base:base.py:409 Calling CTAP2 get_assertion
```
</details>

robin-nitrokey, 2023-04-16:
It looks like this is caused by https://github.com/Nitrokey/trussed/pull/21.  The key is wrapped here with the `rp_id_hash` as associated data:

https://github.com/solokeys/fido-authenticator/blob/0.1.1/src/ctap2.rs#L323-328

But it is unwrapped with empty associated data:

https://github.com/solokeys/fido-authenticator/blob/0.1.1/src/ctap2.rs#L1462-L1480

In old firmware versions, the associated data was ignored when wrapping a key.  Since trussed v0.1.0-nitrokey.9, this is no longer the case.  So we can unwrap old credentials without associated data but fail to unwrap new credentials with associated data.  The easiest fix would be to change `fido-authenticator` to wrap with empty associated data.

sosthene-nitrokey, 2023-04-17:
https://github.com/solokeys/fido-authenticator/pull/24

There's already [this PR](https://github.com/solokeys/fido-authenticator/pull/24) implementing the required fix.

sosthene-nitrokey, 2023-04-17:
Fixed by #236

closed 2023-04-17 by robin-nitrokey