Nitrokey/nitrokey-3-firmware #245: piv and gpg card app used at the same time

apatard, 2023-04-21:
_I'm not sure if it's better to report here or open a ticket to nk support. Since I'm trying things with v1.3.1-test.20230417 firmware, I bet it's better here._

Since there's a PIV app and opengpg app card, I was wondering if I could use both card app at the same time. For instance,
- gpg app for signing/encrypting/auth (which means all 3 slots are used)
- PIV for cases where I need a certificate / pkcs11 support.

I've tested on what's going to be Debian 12, in order to get newer enough libccid and gpg scdaemon. The gnupg scdaemon is configured with:
```
disable-ccid
pcsc-shared
pcsc-driver /usr/lib/x86_64-linux-gnu/libpcsclite.so.1
```

Then:
```
# pkcs11-tool --list-slots
Available slots:
Slot 0 (0x0): Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
  token label        : PIV_II
  token manufacturer : piv_II
  token model        : PKCS#15 emulated
  token flags        : login required, rng, token initialized, PIN initialized
  hardware version   : 0.0
  firmware version   : 0.0
  serial num         : 
  pin min/max        : 4/8
```
but on gpg side:
```
# gpg --card-status
gpg: OpenPGP card not available: Not supported
# gpg-connect-agent 
> LEARN --sendinfo
ERR 100663356 Not supported <SCD>
> scd learn
S READER Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
S SERIALNO ....
INQUIRE KNOWNCARDP  ...
ERR 67109141 IPC call has been cancelled <GPG Agent>
```

I guess that the first command activates the PIV app, which leads to the error with the second command, since the gpg app is not working.

Any hope to get this working ?

robin-nitrokey, 2023-04-21:
Unfortunately, GnuPG requires exclusive access to the smartcard.  So it is not possible to use GnuPG and some other OpenPGP or PIV application in parallel.  You should be able to access them sequentially though by restarting `scdaemon` between calls, e. g. `gpg-connect-agent 'SCD KILLSCD' /bye`.

At some point, we might move the applications to separate card interfaces but there is no ETA for this yet.  As a workaround, you could look into using a different OpenPGP implementation, e. g. sequoia PGP and their [openpgp-card tools](https://gitlab.com/openpgp-card/openpgp-card).

apatard, 2023-04-21:
> Unfortunately, GnuPG requires exclusive access to the smartcard. So it is not possible to use GnuPG and some other OpenPGP or PIV application in parallel. You should be able to access them sequentially though by restarting `scdaemon` between calls, e. g. `gpg-connect-agent 'SCD KILLSCD' /bye`.
> 
Actually, not the case anymore since gnupg 2.2.38 and 2.3.0. There's a new option pcsc-shared to share between gpg scdaemon and other tool.

robin-nitrokey, 2023-04-21:
Hmm, the last time I tried to use that option it did not really work for me.  While technically the pcsc connection was non-exclusive, scdaemon still made the assumption that no other application changes the state of the card during its runtime.  It would be great if this is fixed now.

apatard, 2023-04-21:
I'm currently looking at gnupg git repo, and some commits like https://github.com/gpg/gnupg/commit/044e5a3c3801fde7fa3b26986825377016e73103, https://github.com/gpg/gnupg/commit/f7e00dc73dd0a71e21a30e2b4393a6865ef05718, https://github.com/gpg/gnupg/commit/fa4f716917e545a98e7c22469c51e980aa79443d, there's a small of hope it may work some day, as long as checks on yubikeys are extended.

apatard, 2023-04-25:
I've written a quick&dirty patch on top of gnupg 2.4 (with commits f7e00dc73dd0a71e21a30e2b4393a6865ef05718 and fa4f716917e545a98e7c22469c51e980aa79443d). It seems to be working enough to allow doing "gpg --card-status && pkcs11-tool --list-slots && gpg --card-status". I've not done more testing.

Even if I clean this patch, the following points are remaining:
- not sure if the changes are 100% fine regarding the SELECT command used to switch apps. The code comment is talking about some specific behaviour of yubikey. So, it's working with nk3 maybe by luck ?
- the way to detect the nitrokey is done by checking that the ATR starts with "Nitrokey", which is obvsiouly fragile,
- the gnupg devs may not want to extend the yubikey support to other devices ?
- Debian will ship 2.2.x in bookworm I guess and this patch is working only on 2.4. I fear 2.2.x branch is missing a lot of code to be able to backport the change.

imho, things would be easier if there were different interfaces as it was suggested in https://github.com/Nitrokey/nitrokey-3-firmware/issues/245#issuecomment-1517903749

robin-nitrokey, 2023-04-25:
Nice!  Can you share your patch for reference?  I think this would not even have to be Nitrokey-specific.  Having multiple applications on a card is a standard functionality and should be supported by GnuPG out of the box.  (The difference to the Yubikey is that GnuPG itself supports custom Yubikey commands.)  If GnuPG does not want to change the default behavior, maybe there could be a flag similar to `--pcsc-shared` to enable a multi-application mode that always calls `SELECT` before issuing a command.

apatard, 2023-04-26:
I guess it should be nitrokey specific, as it may later allow (if we want) add nk3 specifics, for instance in the gpg-card tool.
I've put the current patch here: https://gist.github.com/apatard/2049a50505a482c3e1e7906cba40fd72
It's against 2.4 with f7e00dc73dd0a71e21a30e2b4393a6865ef05718 and fa4f716917e545a98e7c22469c51e980aa79443d. Should apply on git head.
IMHO, the changes related to maybe_check_aid are obviously hacks