Nitrokey/nitrokey-3-firmware #250: v1.3.1-test.20230417 : generator didn't stop after throw()

alci63, 2023-04-24:
Hi,

I just received my nk 3C NFC, so I thought I will play with it and try to get the openpgp app working.
As I understand it, v1.3.1-test.20230417 does include the openpgp app. But installing fails:

```
Command line tool to interact with Nitrokey devices 0.4.36
Do you want to download the firmware version v1.3.1-test.20230417? [Y/n]: 
Download v1.3.1-test.20230417: 100%|███████████████████████████████████████| 918k/918k [00:00<00:00, 6.09MB/s]
Current firmware version:  v1.3.1
Updated firmware version:  v1.3.1-test.20230417

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Critical error:
An unhandled exception occurred
        Exception encountered: RuntimeError("generator didn't stop after throw()")

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.g56tsk95' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

robin-nitrokey, 2023-04-24:
Can you please share the full log (`/tmp/nitropy.log.g56tsk95`)?

We had a user reporting a similar issue recently (https://github.com/Nitrokey/pynitrokey/issues/368).  In that case, it was fixed by booting into bootloader manually:
```
$ nitropy nk3 reboot --bootloader
$ nitropy nk3 list
# output should include Nitrokey 3 Bootloader (LPC55) followed by the UUID
$ nitropy nk3 update
```

alci63, 2023-04-25:
Full log joined of a failed attempt:
[nitropy.log.lgksqfe2.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/11324459/nitropy.log.lgksqfe2.txt)

And booting into bootloader manually did the trick for me too:

❯  nitropy nk3 reboot --bootloader
Command line tool to interact with Nitrokey devices 0.4.36
Please press the touch button to reboot the device into bootloader mode ...
 
❯  nitropy nk3 list
Command line tool to interact with Nitrokey devices 0.4.36
:: 'Nitrokey 3' keys
/dev/hidraw6: Nitrokey 3 Bootloader (LPC55) DEEEB98970A1805BB72D80F947F17D77

❯  nitropy nk3 update --version v1.3.1-test.20230417
Command line tool to interact with Nitrokey devices 0.4.36
Do you want to download the firmware version v1.3.1-test.20230417? [Y/n]: 
Download v1.3.1-test.20230417: 100%|██████████████████████████████████████████████████████████████████████| 918k/918k [00:00<00:00, 5.53MB/s]
Current firmware version:  [unknown]
Updated firmware version:  v1.3.1-test.20230417

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y
Perform firmware update: 100%|████████████████████████████████████████████████████████████████████████████| 463k/463k [00:05<00:00, 89.1kB/s]
Finalize upgrade: 100%|█████████████████████████████████████████████████████████████████████████████████████| 100/100 [00:01<00:00, 59.60%/s]

alci63, 2023-04-27:
Sure, I will test any patch.

robin-nitrokey, 2023-04-27:
Great!  I’ve prepared a patch [here](https://github.com/Nitrokey/pynitrokey/pull/378).  It should at least improve the error message but hopefully also fixes the problem.  You can try it with:

```
$ pipx install --suffix=-git-378 git+https://github.com/Nitrokey/pynitrokey.git@refs/pull/378/head
$ nitropy-git-378 nk3 update
$ pipx uninstall pynitrokey-git-378
```

alci63, 2023-04-27:
So, after manually booting in bootloader two days ago, I was able to install v1.3.1-test.20230417.
Test firmware was installed, and gpg was able to ineract with the key.
But then, 'nitropy nk3 update' was not able to get me back to standard 1.3.1 version.

Then I tested your patch:
```
$ pipx install --suffix=-git-378 git+https://github.com/Nitrokey/pynitrokey.git@refs/pull/378/head
$ nitropy-git-378 nk3 update
```

This seems to works, and get me back to 1.31:

```
Do you want to download the firmware version v1.3.1? [Y/n]: 
Download v1.3.1: 100%|█████████████████████████████████████████████████████| 625k/625k [00:00<00:00, 6.46MB/s]
Current firmware version:  [unknown]
Updated firmware version:  v1.3.1

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y
Perform firmware update: 100%|█████████████████████████████████████████████| 316k/316k [00:04<00:00, 76.8kB/s]
Finalize upgrade: 100%|██████████████████████████████████████████████████████| 100/100 [00:06<00:00, 14.68%/s]

```
then back to Test fails:
```
❯  nitropy-git-378 nk3 update --version v1.3.1-test.20230417
Command line tool to interact with Nitrokey devices 0.4.36
Do you want to download the firmware version v1.3.1-test.20230417? [Y/n]: 
Download v1.3.1-test.20230417: 100%|███████████████████████████████████████| 918k/918k [00:00<00:00, 6.16MB/s]
Current firmware version:  v1.3.1
Updated firmware version:  v1.3.1-test.20230417

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Critical error:
An unhandled exception occurred
        Exception encountered: RuntimeError("generator didn't stop after throw()")

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.56d3uwrn' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

or finally works if I retry:

```
❯  nitropy-git-378 nk3 update --version v1.3.1-test.20230417
Command line tool to interact with Nitrokey devices 0.4.36
Do you want to download the firmware version v1.3.1-test.20230417? [Y/n]: 
Download v1.3.1-test.20230417: 100%|███████████████████████████████████████| 918k/918k [00:00<00:00, 6.11MB/s]
Current firmware version:  [unknown]
Updated firmware version:  v1.3.1-test.20230417

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y
Perform firmware update: 100%|█████████████████████████████████████████████| 463k/463k [00:05<00:00, 81.4kB/s]
Finalize upgrade: 100%|█████████████████████████████████████████████████████████████████████| 100/100 [00:06<00:00, 14.62%/s]
```

So, it kinda works :-)
Regards,
Franck

APN-Pucky, 2023-04-28:
I ran into the same problem. After multiple bootloader reboot runs causing
`Exception encountered: McuBootConnectionErrror()` (needs to by plugging in again then).
What seemed to solve it for me is that I experimented with different features (PAM, NFC, U2F).
Then suddenly the flashing worked.
The test after the first flash gave:
```
[3/4]	status   	Device status           	FAILURE  	init error: EXTERNAL_FLASH_ERROR (0x4)
```
But flashing again fixed that (or it just needed to be unplugged once).
I would guess it is a software (dependency?) issue since 2 keys worked on the same machine for me but not on another (both tested `pipx` with `0.4.35` (this worked in the end, on one machine) and `0.4.36` ). However I still could not get the flashing to work on the second PC.
Hope this somehow helps resolving the issue.

runcom, 2023-05-11:
Ran into the very same issue now updating an nk3a nfc, ran the update command once more to make it through the upgrade

```
➜  ~ nitropy nk3 update
Command line tool to interact with Nitrokey devices 0.4.35
Do you want to download the firmware version v1.4.0? [Y/n]: Y
Download v1.4.0: 100%|███████████████████████████████████████████████| 894k/894k [00:00<00:00, 4.72MB/s]
Current firmware version:  v1.3.1-test.20230417
Updated firmware version:  v1.4.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Critical error:
An unhandled exception occurred
	Exception encountered: RuntimeError("generator didn't stop after throw()")

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.wnp_q0kc' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting

➜  ~ nitropy nk3 update
Command line tool to interact with Nitrokey devices 0.4.35
Do you want to download the firmware version v1.4.0? [Y/n]: 
Download v1.4.0: 100%|███████████████████████████████████████████████| 894k/894k [00:00<00:00, 4.94MB/s]
Current firmware version:  [unknown]
Updated firmware version:  v1.4.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y
Perform firmware update: 100%|███████████████████████████████████████| 451k/451k [00:08<00:00, 51.9kB/s]
Finalize upgrade: 100%|████████████████████████████████████████████████| 100/100 [00:01<00:00, 85.47%/s]
```

szszszsz, 2023-05-12:
@robin-nitrokey This looks more like a pynitrokey issue. Shall we move it there?

robin-nitrokey, 2023-05-23:
Moved to `pynitrokey`: https://github.com/Nitrokey/pynitrokey/issues/394

closed 2023-05-23 by robin-nitrokey