Nitrokey/nitrokey-3-firmware #254: ssh-keygen with resident ed25519 key does not work

robin-nitrokey, 2023-04-28:
Triggered by the `test_ssh_resident[ed25519]` test case from https://github.com/Nitrokey/nitrokey-3-tests/commit/41be99d4325bb466bd16a3c2bbd6d0e4da8282ea:

```
E           pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
E           <pexpect.pty_spawn.spawn object at 0x7ff621f7fb20>
E           command: /usr/bin/ssh-keygen
E           args: ['/usr/bin/ssh-keygen', '-t', 'ed25519-sk', '-f', '/tmp/tmp7gb0zjeh/ed25519-sk', '-C', 'fido', '-P', '', '-O', 'resident']
E           buffer (last 100 chars): b''
E           before (last 100 chars): b': \r\nKey enrollment failed: invalid format\r\r\n'
```

<details>
<summary>usbip runner log</summary>

```
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > FF FF FF FF 86 00 08 52 42 F4 F8 30 85 5F 94 00
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 90 00 01 04 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  !                                                > found aaguid
 INFO  usbd_ctaphid::pipe                               > Got 159 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 90 00 01 04 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  !                                                > found aaguid
 INFO  usbd_ctaphid::pipe                               > Got 159 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 90 00 06 06 A2 01 01 02 02 00 00 00
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbd_ctaphid::pipe                               > Got 81 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 90 00 BC 06 A6 01 01 02 04 03 A5 01
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 00 57 15 4C 2B 14 EE A7 A5 E0 66 96
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 01 BE AE 7B 4D 37 D6 6B AA 70 27 98
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 13 02 80 24 8C 80 53 95 76 F2 1A 0E AC
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
[... 100 lines cut ...]
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  !                                                > this may be an RK: Some(DirEntry { file_name: p"82fa7e99d1143352\0", metadata: Metadata { file_type: File, size: 129 }, path: p"rk/e30610e8a1621159/82fa7e99d1143352\0" })
 INFO  !                                                > checking RK p"rk/e30610e8a1621159/82fa7e99d1143352\0" for userId 
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  !                                                > :: deleting resident key
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  !                                                > Overwriting previous rk tied to this userId.
 INFO  fido_authenticator::ctap2::credential_management > get metadata
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbip_runner                                     > Set status: Processing
 INFO  usbip_runner                                     > Set status: Idle
 INFO  usbd_ctaphid::pipe                               > Got 1 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 14 91 00 00 00 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe                               > init
```
</details>

sosthene-nitrokey, 2023-04-28:
Similar error with ecdsa:

```
>ssh-keygen -t ecdsa-sk -O resident -f test4.key
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
Enter PIN for authenticator: 
You may need to touch your authenticator again to authorize key generation.
Key enrollment failed: invalid format
```

usbip logs: 

```
>RUST_LOG=debug,usbip_device=off cargo run --release -- --ifs fido_efs
    Finished release [optimized + debuginfo] target(s) in 0.10s
warning: the following packages contain code that will be rejected by a future version of Rust: bindgen v0.56.0, nom v5.1.2
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 202`
     Running `/home/soso-nitrokey/Code/nitrokey-3-firmware/target/release/usbip-runner --ifs fido_efs`
 INFO  usbip_runner > Initializing Trussed
 INFO  trussed_usbip > Ready for work
 INFO  usbd_ctaphid::pipe > >> 
 INFO  usbd_ctaphid::pipe > FF FF FF FF 86 00 08 40 A0 A7 68 80 68 5A 34 00
 INFO  usbd_ctaphid::pipe > init
 INFO  usbd_ctaphid::pipe > >> 
 INFO  usbd_ctaphid::pipe > 00 00 00 01 90 00 01 04 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe > init
 DEBUG !                  > ctaphid-dispatch: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > handle CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > 1a SP: 20000000
 DEBUG trussed_usbip                > syscall
 INFO  usbip_runner                 > Set status: Processing
 DEBUG !                            > checking existence of trussed/dat/rng-state.bin
 DEBUG !                            > reading trussed/dat/rng-state.bin
 DEBUG !                            > storing trussed/dat/rng-state.bin
 DEBUG !                            > writing trussed/dat/rng-state.bin
 DEBUG !                            > TRUSSED ReadFile(ReadFile { location: Internal, path: p"persistent-state.cbor\0" })
 DEBUG !                            > reading fido/dat/persistent-state.cbor
 INFO  usbip_runner                 > Set status: Idle
 DEBUG !                            > I/E/V : 107/30/126 >
 INFO  fido_authenticator::state    > loaded previous state!
 DEBUG fido_authenticator::dispatch > try_handle CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > try_get CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > 2a SP: 20000000
 DEBUG !                            > remaining stack size: 0 bytes
 DEBUG trussed_usbip                > syscall
 INFO  usbip_runner                 > Set status: Processing
 DEBUG !                            > TRUSSED Exists(Exists { mechanism: P256, key: KeyId(Id(00)) })
 INFO  usbip_runner                 > Set status: Idle
 DEBUG !                            > I/E/V : 107/30/126 >
 INFO  !                            > attestation key does not exist
 DEBUG fido_authenticator::dispatch > end handle CTAP2
 INFO  usbd_ctaphid::pipe           > Got 159 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe           > >> 
 INFO  usbd_ctaphid::pipe           > 00 00 00 01 90 00 01 04 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe           > init
 DEBUG !                            > ctaphid-dispatch: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > handle CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > 1a SP: 20000000
 DEBUG fido_authenticator::dispatch > try_handle CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > try_get CTAP2: remaining stack: 0 bytes
 DEBUG fido_authenticator::dispatch > 2a SP: 20000000
 DEBUG !                            > remaining stack size: 0 bytes
 DEBUG trussed_usbip                > syscall
 INFO  usbip_runner                 > Set status: Processing
 DEBUG !                            > TRUSSED Exists(Exists { mechanism: P256, key: KeyId(Id(00)) })
 INFO  usbip_runner                 > Set status: Idle
 DEBUG !                            > I/E/V : 107/30/126 >
 INFO  !                            > attestation key does not exist
 DEBUG fido_authenticator::dispatch > end handle CTAP2
[... 486 lines cut ...]
 DEBUG !                              > TRUSSED ReadDirFirst(ReadDirFirst { location: Internal, dir: p"rk/e30610e8a1621159\0", not_before_filename: None })
 INFO  usbip_runner                   > Set status: Idle
 DEBUG !                              > I/E/V : 107/30/120 >
 INFO  !                              > this may be an RK: None
 INFO  fido_authenticator::ctap2::credential_management > get metadata
 DEBUG trussed_usbip                                    > syscall
 INFO  usbip_runner                                     > Set status: Processing
 DEBUG !                                                > TRUSSED ReadDirFirst(ReadDirFirst { location: Internal, dir: p"rk\0", not_before_filename: None })
 INFO  usbip_runner                                     > Set status: Idle
 DEBUG !                                                > I/E/V : 107/30/120 >
 INFO  fido_authenticator::ctap2::credential_management > First return
 INFO  fido_authenticator::ctap2::credential_management > Counting all
 DEBUG trussed_usbip                                    > syscall
 INFO  usbip_runner                                     > Set status: Processing
 DEBUG !                                                > TRUSSED ReadDirFirst(ReadDirFirst { location: Internal, dir: p"rk/e30610e8a1621159\0", not_before_filename: None })
[... cut ...]
 DEBUG fido_authenticator::dispatch                     > end handle CTAP2
 INFO  usbd_ctaphid::pipe                               > Got 1 bytes response from authenticator, starting send
 INFO  usbd_ctaphid::pipe                               > >> 
 INFO  usbd_ctaphid::pipe                               > 00 00 00 02 91 00 00 00 00 00 00 00 00 00 00 00
 INFO  usbd_ctaphid::pipe                               > init
 INFO  usbd_ctaphid::pipe                               > Got waiting reply from authenticator??
```

robin-nitrokey, 2023-04-28:
The problem is:

- Credentials are stored in one directory per Relying Party (RP).
- When registering a new resident key, we delete existing resident keys with the same RP + user ID.
- If there is such a credential but no other credential for that RP, this means that the RP directory is now empty.
- We then call `CredentialManagement::get_creds_metadata` to determine if we have reached the maximum credential count
- This method does not expect empty RP dirs, causing the issue.

Suggested fix:
1. Allow empty directories in `CredentialManagement::get_creds_metadata`.
2. (optional) Delete empty RP directory if the credential limit is reached.  This is only relevant for the case where the limit was exceeded before the checks were introduced.

closed 2023-05-02 by robin-nitrokey