Nitrokey/nitrokey-3-firmware #259: Nitrokey 3C Firmware v1.4.0-rc.2 FIDO2 Error

Cthulu201, 2023-05-04:
**tl;dr:**
Just posting test results and the associated log file for tracking:

I have 2 Nitrokeys (3A and 3C), and I was testing the beta firmware on one of them (3C) - upgraded from 1.3.1 to 1.4.0-rc.2. I tried using sudo and just user credentials and get the same results. I know it's beta, so I didn't expect a polished firmware. I'm using Arch Linux. I don't know why the status says no key found but the test still runs.

```
#
# nitropy status
#

❯ nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.36
Critical error:
No Nitrokey 3 device found

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.3iooq4bs' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

```
#
# nitropy test
#

❯ nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/5]	uuid         	UUID query              	SUCCESS  	C150411AFFFBD551967B49EFB4A1205D
[2/5]	version      	Firmware version query  	SUCCESS  	v1.4.0-rc.2
[3/5]	status       	Device status           	SUCCESS  	Status(init_status=<InitStatus: 0>, ifs_blocks=48, efs_blocks=474)
[4/5]	provisioner  	Firmware mode           	SUCCESS  	
Please press the touch button on the device ...
[5/5]	fido2        	FIDO2                   	FAILURE  	PIN activated -- please set the --pin option

5 tests, 4 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.tisqh339' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

```
#
# nitropy log file
#

❯ cat /tmp/nitropy.log.tisqh339
135        INFO pynitrokey.cli Timestamp: 2023-05-04 15:41:34.305281
135        INFO pynitrokey.cli OS: uname_result(system='Linux', node='randsum', release='6.3.1-arch1-1', version='#1 SMP PREEMPT_DYNAMIC Mon, 01 May 2023 17:42:39 +0000', machine='x86_64')
135        INFO pynitrokey.cli Python version: 3.11.3
137        INFO pynitrokey.cli pynitrokey version: 0.4.36
138        INFO pynitrokey.cli cryptography version: 40.0.2
[... 84 lines cut ...]
573       DEBUG       root print: Summary: 1 device(s) tested, 0 successful, 1 failed
573       DEBUG       root print: Critical error:
573       DEBUG       root print: Test failed for 1 device(s)
573       DEBUG       root listing all connected devices:
582       DEBUG       root :: 'Nitrokey FIDO2' keys
582       DEBUG       root :: 'Nitrokey Start' keys:
613       DEBUG       root :: 'Nitrokey 3' keys
616        INFO  libusbsio HID enumeration[94525200757648]: initialized
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #0: Nitrokey 3
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #1: LED Controller
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #2: U2417H_0B29_15083001
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #3: HP Link-5 Micro Receiver
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #4: HP Link-5 Micro Receiver
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #5: Kensington Expert Mouse
616       DEBUG  libusbsio HID enumeration[94525200757648]: device #6: U2417H_0B29_15083001
616        INFO  libusbsio HID enumeration[94525200757648]: finished, total 7 devices
642       DEBUG       root /dev/hidraw0: Nitrokey 3 C150411AFFFBD551967B49EFB4A1205D
642       DEBUG       root print: --------------------------------------------------------------------------------
642       DEBUG       root print: Critical error occurred, exiting now
642       DEBUG       root print: Unexpected? Is this a bug? Would you like to get support/help?
642       DEBUG       root print: - You can report issues at: https://support.nitrokey.com/
642       DEBUG       root print: - Writing an e-mail to support@nitrokey.com is also possible
642       DEBUG       root print: - Please attach the log: '/tmp/nitropy.log.tisqh339' with any support/help request!
642       DEBUG       root print: - Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

robin-nitrokey, 2023-05-04:
What exactly is the error you see?  The `nitropy nk3 test` failure seems to be expected because there is a PIN set:

```
[5/5]	fido2        	FIDO2                   	FAILURE  	PIN activated -- please set the --pin option
```

Does it work if you use the `--pin` option?  Or did you not set a PIN?

Cthulu201, 2023-05-04:
On v1.3.1 I set the pin per the website via chrome://settings/securityKeys. After upgrading to v1.4.0-rc.2, when I run the test with the new firmware, where it's supposed to let me press, the LED flashes 1X and then the rest of the test prints out. I don't even get a chance to press.

Cthulu201, 2023-05-04:
```
❯ nitropy fido2 set-pin
Command line tool to interact with Nitrokey devices 0.4.36
Please enter new pin:
Please confirm new pin:
Critical error:
failed setting new pin, maybe it's already set?
to change an already set pin, please use:
$ nitropy fido2 change-pin
        Exception encountered: CtapError('CTAP error: 0x30 - NOT_ALLOWED')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log._jamy7by' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```
```
❯ nitropy fido2 change-pin
Command line tool to interact with Nitrokey devices 0.4.36
Please enter old pin:
Please enter new pin:
Please confirm new pin:
done - please use new pin to verify key
```
```
❯ nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/5]   uuid            UUID query                      SUCCESS         C150411AFFFBD551967B49EFB4A1205D
[2/5]   version         Firmware version query          SUCCESS         v1.4.0-rc.1
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus: 0>, ifs_blocks=48, efs_blocks=474)
[4/5]   provisioner     Firmware mode                   SUCCESS
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           FAILURE         PIN activated -- please set the --pin option

5 tests, 4 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.zxiyflmn' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

robin-nitrokey, 2023-05-04:
If you have set a pin, you have to call `nitropy nk3 test --pin <your-pin>` to run the tests.

Cthulu201, 2023-05-04:
RGR, that works
```
❯ nitropy nk3 test --pin 123456
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw1

Running tests for Nitrokey 3 at /dev/hidraw1

[1/5]   uuid            UUID query                      SUCCESS         C150411AFFFBD551967B49EFB4A1205D
[2/5]   version         Firmware version query          SUCCESS         v1.4.0-rc.2
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus: 0>, ifs_blocks=48, efs_blocks=474)
[4/5]   provisioner     Firmware mode                   SUCCESS
Please press the touch button on the device ...
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           SUCCESS

5 tests, 5 successful, 0 skipped, 0 failed

Summary: 1 device(s) tested, 1 successful, 0 failed
```
That works. Just fails on the nitropy nk3 test, which doesn't allow me to touch the Nitrokey either. Guess the nitropy codes has bugs?
```
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           FAILURE         PIN activated -- please set the --pin option
```

Cthulu201, 2023-05-11:
Nitrokey 3C test still doesn't allow me to press the button. It prompts and then goes to 5/5 failure. The PIN has been set.

```
❯ nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/5]   uuid            UUID query                      SUCCESS         C150411AFFFBD551967B49EFB4A1205D
[2/5]   version         Firmware version query          SUCCESS         v1.4.0
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus: 0>, ifs_blocks=48, efs_blocks=474)
[4/5]   provisioner     Firmware mode                   SUCCESS
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           FAILURE         PIN activated -- please set the --pin option

5 tests, 4 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.f2yk3r1p' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

Nitrokey 3A pre-firmware upgrade, no issues
```
~/dev/git
❯ nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/5]   uuid            UUID query                      SUCCESS         6F5B9DEE73F35E54B4743BC2551B1DD2
[2/5]   version         Firmware version query          SUCCESS         v1.3.1
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus: 0>, ifs_blocks=60, efs_blocks=478)
[4/5]   provisioner     Firmware mode                   SUCCESS
Please press the touch button on the device ...
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           SUCCESS

5 tests, 5 successful, 0 skipped, 0 failed

Summary: 1 device(s) tested, 1 successful, 0 failed
```

Nitrokey 3A post-firmware upgrade, no issues. 
```
❯ nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.36
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw0

Running tests for Nitrokey 3 at /dev/hidraw0

[1/5]   uuid            UUID query                      SUCCESS         6F5B9DEE73F35E54B4743BC2551B1DD2
[2/5]   version         Firmware version query          SUCCESS         v1.4.0
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus: 0>, ifs_blocks=60, efs_blocks=478)
[4/5]   provisioner     Firmware mode                   SUCCESS
Please press the touch button on the device ...
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           SUCCESS

5 tests, 5 successful, 0 skipped, 0 failed

Summary: 1 device(s) tested, 1 successful, 0 failed
```

szszszsz, 2023-05-12:
@Cthulu201 Please run the failing test with `nitropy nk3 test --pin <your-pin>`

Cthulu201, 2023-05-12:
Sorry, I guess the results aren't illustrating my point very well.

I have 2 Nitrokey 3s, a 3A and a 3C. The tests, per the Nitrokey website instructions, run fine on the 3A before and after the upgrade (the last 2 results) - and even before.

However, the 3C fails fido2, and, even though it shows the prompt to touch the 3C, it doesn't allow any time before the "[5/5] ... FAILURE"

There isn't anything anywhere, other than this github issue, where it states "Ignore the 3C errors and run the test --pin <pin> option instead." Is the issue in the firmware, does the nitropy code need to be fixed (nitropy fido2 relevant to Nitrokey3?), does the website need to be updated, or D?

szszszsz, 2023-05-15:
> There isn't anything anywhere, other than this github issue, where it states "Ignore the 3C errors and run the test --pin option instead." Is the issue in the firmware, does the nitropy code need to be fixed (nitropy fido2 relevant to Nitrokey3?), does the website need to be updated, or D?

@Cthulu201 
It's there, right in the error message:
> [5/5]   fido2           FIDO2                           FAILURE         PIN activated -- please set the --pin option

Cthulu201, 2023-05-30:
> Ok, trying to summarize (because I am a little confused):
> 
> * Your Nitrokey 3C was working before the update
> * Now you have 1.4 on your NK3 C and the touch button request is canceled without your interaction (while running `nitropy nk3 test --pin <your-pin>`)?
>   Is this correct?

Close.

`nitropy nk3 test` runs without issue on Nitrokey 3A - I can run it with or without `--pin <my_pin>` option without any error.

Nitrokey 3C touch button request is cancelled without my interaction when running `nitropy nk3 test`. It's fine when I run `--pin <my_pin>`

daringer, 2023-06-20:
Okok, well then this is kind of expected:

> Nitrokey 3C touch button request is cancelled without my interaction when running `nitropy nk3 test`.

The output is simply not correct at this point, so this issue properly boils down to: 

* Don't show the user-presence request, if `nk3 test` is called without pin, but a pin is expected

szszszsz, 2023-06-26:
Let's close it here, and make a proper pynitrokey ticket
@daringer @robin-nitrokey

robin-nitrokey, 2023-06-26:
Moved to: https://github.com/Nitrokey/nitrokey-3-firmware/issues/259

closed 2023-06-26 by robin-nitrokey