Nitrokey/nitrokey-3-firmware #262: random crashes

primalmotion, 2023-05-06:
It seems the nitrokey3 "crashes" pretty often (firmware 1.4). When that happens, clients like gpg, or firefox, or PAM, are just "stuck". sometimes the key returns a card reading error, something nothing. 

Unplugging/repluging fixes the issue. I'm not sure where to pull relevant logs from, but this happens around 3 or 4 times a day while using GPG and FIDO2 function, mostly

robin-nitrokey, 2023-05-07:
When the problem happens, do you see a change of the LED color?  What is the result of running `nitropy nk3 list` and `nitropy nk3 status` before replugging the device?

primalmotion, 2023-05-07:
I'll check all of that next time it happens

primalmotion, 2023-05-08:
Just happened while trying FIDO auth. The led is solid green

```
 nitropy nk3 list
Command line tool to interact with Nitrokey devices 0.4.36
:: 'Nitrokey 3' keys
Critical error:
An unhandled exception occurred
	Exception encountered: CtapError('CTAP error: 0x06 - CHANNEL_BUSY')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.kfjcakxn' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

```
nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.36
Critical error:
An unhandled exception occurred
	Exception encountered: CtapError('CTAP error: 0x06 - CHANNEL_BUSY')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.n_r__211' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

primalmotion, 2023-05-08:
Here's the log the command asks me to paste :)

````
284        INFO pynitrokey.cli Timestamp: 2023-05-08 09:05:50.433152
284        INFO pynitrokey.cli OS: uname_result(system='Linux', node='axion', release='6.3.1-arch1-1', version='#1 SMP PREEMPT_DYNAMIC Mon, 01 May 2023 17:42:39 +0000', machine='x86_64')
284        INFO pynitrokey.cli Python version: 3.11.3
293        INFO pynitrokey.cli pynitrokey version: 0.4.36
293        INFO pynitrokey.cli cryptography version: 40.0.2
294        INFO pynitrokey.cli ecdsa version: 0.18.0
294        INFO pynitrokey.cli fido2 version: 1.1.1
294        INFO pynitrokey.cli pyusb version: 1.2.1
295        INFO pynitrokey.cli spsdk version: 1.8.0
634        INFO  libusbsio Loading SIO library: /usr/lib/python3.11/site-packages/libusbsio/bin/linux_x86_64/libusbsio.so
640        INFO  libusbsio HID enumeration[94575114131280]: initialized
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #0: Nitrokey 3
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #1: Device Management
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #2: Trackpad / Boot
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #3: Actuator
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #4: Accelerometer
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #5: KBD67v2
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #6: KBD67v2
640       DEBUG  libusbsio HID enumeration[94575114131280]: device #7: KBD67v2
640        INFO  libusbsio HID enumeration[94575114131280]: finished, total 8 devices
645       DEBUG fido2.hid.linux Failed opening device /dev/hidraw8
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw8'
704       DEBUG fido2.hid.linux Failed opening device /dev/hidraw6
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw6'
705       DEBUG fido2.hid.linux Failed opening device /dev/hidraw5
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw5'
705       DEBUG fido2.hid.linux Failed opening device /dev/hidraw4
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw4'
705       DEBUG fido2.hid.linux Failed opening device /dev/hidraw3
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw3'
706       DEBUG fido2.hid.linux Failed opening device /dev/hidraw2
Traceback (most recent call last):
[... 211 lines cut ...]
  File "/usr/lib/python3.11/site-packages/pynitrokey/nk3/device.py", line 146, in list
    devices.append(Nitrokey3Device(device))
                   ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pynitrokey/nk3/device.py", line 73, in __init__
    self.admin.status()
  File "/usr/lib/python3.11/site-packages/pynitrokey/nk3/admin_app.py", line 71, in status
    reply = self._call(AdminCommand.STATUS)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pynitrokey/nk3/admin_app.py", line 58, in _call
    return self.device._call(
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pynitrokey/nk3/device.py", line 133, in _call
    response = self.device.call(command.value, data=data)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/fido2/hid/__init__.py", line 216, in call
    raise CtapError(struct.unpack_from(">B", recv)[0])
fido2.ctap.CtapError: CTAP error: 0x06 - CHANNEL_BUSY
900       DEBUG       root print: --------------------------------------------------------------------------------
900       DEBUG       root print: Critical error occurred, exiting now
900       DEBUG       root print: Unexpected? Is this a bug? Would you like to get support/help?
900       DEBUG       root print: - You can report issues at: https://support.nitrokey.com/
900       DEBUG       root print: - Writing an e-mail to support@nitrokey.com is also possible
900       DEBUG       root print: - Please attach the log: '/tmp/nitropy.log.n_r__211' with any support/help request!
900       DEBUG       root print: - Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
````

szszszsz, 2023-05-09:
> Just happened while trying FIDO auth.

Hi! Can you share which browser do you use?

primalmotion, 2023-05-09:
Yes, it's FF. What else? :)

sosthene-nitrokey, 2023-05-16:
Given the `CHANNEL_BUSY` error, I guess it's the same error as https://github.com/Nitrokey/nitrokey-3-firmware/issues/200

closed 2023-09-12 by daringer