Nitrokey/nitrokey-3-firmware #263: gpg does not recognized the nitrokey

Chimrod, 2023-05-07:
Hello, I’ve upgraded the nitrokey into the latest firmware, and the operation ran fine:

```
(nitrokey) sebastien@trunk:~$ nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.36
UUID:               B9EF61DC738B0E569C389B06AA249067
Firmware version:   v1.4.0-rc.2
Init status:        ok
Free blocks (int):  60
Free blocks (ext):  478
```

However, the card is not recognized by gpg:

```
(nitrokey) sebastien@trunk:~$ LANG=C gpg --card-status 
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

If I plug a nitrokey start instead the key is identified so this does not seems to be a configuration error from gpg:

```
(nitrokey) sebastien@trunk:~$ LANG=C gpg --card-status 
Reader ...........: Nitrokey Nitrokey Start (FSIJ-1.2.19-43131736) 00 00
Application ID ...: D276000124010200FFFE431317360000
Application type .: OpenPGP
Version ..........: 2.0
```

Am I missing something ?

robin-nitrokey, 2023-05-07:
Is `pcscd` running on your system (i. e. does `pgrep pcscd` show some output)?  If yes, what is the output of `pcsc_scan -r` when the Nitrokey 3 is connected?

Chimrod, 2023-05-07:
Hello, I’ve tried with both the nk3 and the nitrokey start.

With the nk3, the application does not return, just waiting:

```
sebastien@trunk:~$ pcsc_scan -r
Using reader plug'n play mechanism
Scanning present readers...
Waiting for the first reader... /   
```

At the opposite, with the nitrokey start, I have an answer directly:

```
sebastien@trunk:~$ pcsc_scan -r
Using reader plug'n play mechanism
Scanning present readers...
0: Nitrokey Nitrokey Start (FSIJ-1.2.19-43131736) 00 00
```

For your information, the nk3 comes from a pre-order batch from 2021/04

robin-nitrokey, 2023-05-07:
What distribution do you use?  Can you find out the version of `libccid` on your system?

Chimrod, 2023-05-07:
I’m using debian stable. The version of the library is [1.4.34-1](https://packages.debian.org/bullseye/libccid)

robin-nitrokey, 2023-05-07:
Okay, 1.5.0 is the first release of pcscd that supports the Nitrokey 3.  Please try the following steps to fix this problem:
1. Backup the current `/etc/libccid_Info.plist` file.
2. Obtain a current `Info.plist` file:
   - You can download it from [this repository](https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/Info.plist).
   - You can extract it from the Debian package for 1.5.0 or later.
   - You can build it from source:  Download the latest release from https://ccid.apdu.fr/files/ and run `make Info.plist` inside `src`.
3. Make sure that the file contains the string `Nitrokey 3`.
4. Copy the new `Info.plist` file to `/etc/libccid_Info.plist`.
5. Restart `pcscd` and re-run `pcsc_scan -r`.

Chimrod, 2023-05-07:
I’ve backported the version from testing on my PC and this works now.

For the other people using debian stable, here is [the documentation I’ve followed](https://wiki.debian.org/SimpleBackportCreation) and the steps I’ve used for that:

```
mkdir backports && cd backports/
apt source ccid
cd ccid-1.5.2/
sudo mk-build-deps --install --remove
dch --bpo
fakeroot debian/rules binary
dpkg-buildpackage -b -us -uc
cd ..
sudo apt install ./libccid_1.5.2-1~bpo11+1_amd64.deb
```

And now the key is recognized by gpg:

```
sebastien@trunk:~$ LANG=C gpg --card-status
Reader ...........: Nitrokey Nitrokey 3 [CCID/ICCD Interface] 00 00
Application ID ...: D276000124010304000FB9EF61DC0000
Application type .: OpenPGP
Version ..........: 3.4
```

Thanks for your help, the issue is solved!

closed 2023-05-07 by Chimrod

Artiom-M, 2023-05-09:
I have the same problem. With latest ccid pcsc_scan detects the key, but not  gpg --card-status