Nitrokey/nitrokey-3-firmware #264: Critical error: RuntimeError("generator didn't stop after throw()")

Eragoneq, 2023-05-11:
Hello,

I seem to have a similar problem to #250, where it seems that my device cannot go into bootloader mode. I've tried to reboot manually to bootloader, but it seems to have exactly the same problem as the device is not seen after reboot. 

Here's the log of the update command https://gist.github.com/Eragoneq/0c0f79b113db39cbce9999de585da8c6

```
Command line tool to interact with Nitrokey devices 0.4.36
Do you want to download the firmware version v1.4.0? [Y/n]:
Download v1.4.0: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 894k/894k [00:00<00:00, 5.12MB/s]
Current firmware version:  v1.2.0
Updated firmware version:  v1.4.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Critical error:
An unhandled exception occurred
	Exception encountered: RuntimeError("generator didn't stop after throw()")

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.ft5aayir' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting

Exception ignored in: <generator object Updater._get_bootloader at 0x7f2f02f7fdf0>
RuntimeError: generator ignored GeneratorExit
```

robin-nitrokey, 2023-05-11:
Can you share the output of the update if you call `nitropy nk3 reboot --bootloader` before it?  If the bootloader is not recognized, you might be missing the udev rules: https://docs.nitrokey.com/software/nitropy/linux/udev

Eragoneq, 2023-05-11:
Yeah, it seems that I was missing the rules. 

The update seems to have been successful, but now it also displays the free blocks as it was not doing it before, so is that intended?
```
Command line tool to interact with Nitrokey devices 0.4.36
UUID:               FF3B676168A42C57A70683CF1285EBBE
Firmware version:   v1.4.0
Init status:        ok
Free blocks (int):  46
Free blocks (ext):  478
```

Eragoneq, 2023-05-11:
Okay, after using the `test` command everything seems to be alright, so I will close the issue. Not sure if the nitrokey-app was supposed to install the udev rules, but it also didn't work with it. I suppose there should possibly be some link to it in the app page as well.

closed 2023-05-11 by Eragoneq

robin-nitrokey, 2023-05-11:
> The update seems to have been successful, but now it also displays the free blocks as it was not doing it before, so is that intended?

Yes, this is an additional debugging information.  You don’t have to worry about it.  If the block count would be very low, `nitropy nk3 test` would issue a warning.