Nitrokey/nitrokey-3-firmware #267: Unable to use KDF with Nitrokey 3A Mini and v1.4.0 firmware

ynezz, 2023-05-13:
Thanks a lot for such a nice stack! BTW I was just preparing our Nitrokey 3A Mini keys using following guide:

### Using following environment

 * Fedora 38 container
 * gpg (GnuPG) 2.4.0
 * libgcrypt 1.10.2-unknown

### Install nitropy toolkit

```
  $ python3.10 -m venv venv
  $ source venv/bin/activate
  $ pip install -U pip
  $ pip install pipx
  $ pipx install nitropy
  $ pipx list
  ...
  pynitrokey 0.4.36, installed using Python 3.10.6
```

### Upgrade firmware to v1.4.0

```
 $ nitropy nk3 update --version v1.4.0
 $ nitropy nk3 status
 ...
 Firmware version:   v1.4.0
```

### Perform factory reset and enable KDF-DO

```
 $ gpg --card-edit
 gpg/card> admin
 gpg/card> factory-reset
 gpg/card> kdf-setup
```

  (default admin pin is 12345678)

 ```
 gpg/card> list
 ...
 KDF setting ......: on
 ```

### Prepare PINs

 * Gather some entropy from nk3

```
  $ nk3 rng --length 4096 | sudo tee /dev/random | hexdump -C
```

 * Generate nk3 PIN, Admin PIN and Reset PIN

```
  $ for k in $(seq 1 3); do gpg --gen-random --armor 0 60; sleep 60; done
```
### Prepare key

```
  gpg/card> key-attr
  Set all to ECC / Curve 25519

  gpg/card> forcesig

  gpg/card> name
  Cardholder's surname: Build System Key 3.
  Cardholder's given name: OpenWrt

  gpg/card> url 
  URL to retrieve public key: https://git.openwrt.org/?p=keyring.git;a=blob_plain;f=gpg/CAE438715492B555.asc;hb=HEAD

  gpg/card> passwd
  1 - change PIN 
  4 - set the Reset Code
  3 - change Admin PIN
```

Then when using `keytocard` admin password doesn't work anymore. I've tried the same procedure with Yubikey 5C Nano and it worked fine. As I don't have a spare nk3 (and don't keep GPG master key), I'm not able to debug this further now, going to order some spare keys and would provide more details in the upcoming days hopefully.

sosthene-nitrokey, 2023-05-15:
Hi, thanks for the report.

Admin PINs *must* be larger than 8 bytes (characters). After some testing it appears that the functionality works properly with PINs larger than 8 bytes, but if you change the admin PIN to one less than 8 bytes, GPG will perform the change, but then refuses to check it even if the device would not actually reject it.

Did you try with an admin PIN that is long enough?

sosthene-nitrokey, 2023-05-15:
After more investigation I did find a bug that you may have triggered, fixed in https://github.com/Nitrokey/opcard-rs/pull/158

What is surprising is that it should have affected the user pin, not the admin pin.

I was not able to reproduce the specific issue you encountered without using an admin PIN longer than 8 bytes though.

ynezz, 2023-05-15:
> Note that your command gives a 3 80 bytes ascii strings. 

Thanks, noted, I didn't counted that, correct :)

> Are you using this

I'm using the output directly, copy&pasting it.

sosthene-nitrokey, 2023-05-15:
> Well, I'll reproduce it once I've spare nk3, but IIRC then it was a PIN after keytocard command, so probably admin PIN?

It depends. Depending on the configuration `gpg-agent` can cache the admin PIN for a while, but it should not require the user PIN.

Does using the Admin pin for operations like `name` work?

Is it an issue that has happened every time you tried, or have you tried only once?

szszszsz, 2023-05-15:
Just retested on Nitrokey 3 AM 1.4.0, gpg (GnuPG) 2.4.0, and it works for me.
@ynezz Can you verify please if there were any different steps taken in the log?
[gpg.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/11478965/gpg.txt)

ynezz, 2023-05-18:
Thanks a lot for double checking. I've received the spare keys and checked the procedure again, working fine this time. Sorry for the noise.