Nitrokey/nitrokey-3-firmware #268: nk3 C nfc doesn't support pin (?)

runcom, 2023-05-14:
With systemd-cryptenroll I get the following:

```
Requested to lock with PIN, but FIDO2 device /dev/hidraw0 does not support it, disabling.
```

the A versions of my other keys just work fine afaict and I can enroll with pin - also, the nk3 C nfc is configured the same (same admin/user pin too).

runcom, 2023-05-15:
oddly enough, the first time it didn't ask for the pin on boot, the second time after re-enrolling with the same warning it's now asking the pin :man_shrugging:

runcom, 2023-05-24:
it's some sort of timing/race issue I think - today I've rebooted my laptop and it just didn't ask for the pin - just waited to touch the key and it unlocked, any idea?

runcom, 2023-07-20:
I can confirm that 1.5.0 on a brand new nitrokey 3 NFC C has the issue - randomly either asks the pin or it doesn't and it just requires the touch :shrug: anything I can help with? I'm on Fedora 38

robin-nitrokey, 2024-04-24:
Generally, in FIDO2 the platform (in this case systemd) can determine whether a PIN is required for an operation, not the authenticator (the Nitrokey 3).  If you want to enforce that a credential can only be used with the PIN, you can use the `credProtect` extension when creating the credential.  To me this looks like `systemd-cryptenroll` does not use the `credProtect` extension, and for some reason does not always require a PIN when accessing the credential.

So if the PIN is not queried during boot, it should be investigated what triggeres the PIN query in systemd.

robin-nitrokey, 2025-08-08:
Closing as the problem is not caused by the NK3.

closed 2025-08-08 by robin-nitrokey