Nitrokey/nitrokey-3-firmware #270: NK3A doesn't connect in OpenKeychain on latest firmware

proninyaroslav, 2023-05-18:
NK3A worked with OpenKeychain over NFC on alpha versions 1.3.0 when I tested it in March. I decided to try the same on 1.4.0, and it doesn't connect durin encryption, it says that I removed the token too early, although this is not the case. At the same time, a red light is on on the token.

szszszsz, 2023-05-18:
Connected: https://github.com/Nitrokey/opcard-rs/issues/157

sosthene-nitrokey, 2023-05-22:
OpenPGP is not supported over NFC. Does this happen with FIDO or is it limited to OpenPGP?

ciropom, 2023-05-23:
I have this issue too. 
I created three subkeys (for auth, signing, and encryption) from my primary key in the NC3 NFC,
now I'm trying to import these keys in openkeychain but it says "token removed too early"
no blinking light on my side

ciropom, 2023-05-26:
I'm using NFC. If I click "use token" and then plug the device nothing happens.

ciropom, 2023-05-26:
I'm using /e/ OS on Fairphone 4.

sosthene-nitrokey, 2023-05-26:
I don't have the hardware to actually test it from a phone, but I was able to test it from waydroid. It doesn't appear that OpenKeychain even tries to connect to the device as no logs shown on the nitrokey side.

It looks like OpenKeychain has an allowlist for accepted devices (see [here](https://github.com/open-keychain/open-keychain/blob/master/OpenKeychain/src/main/java/org/sufficientlysecure/keychain/securitytoken/UsbConnectionDispatcher.java#L95) and [here](https://github.com/open-keychain/open-keychain/blob/8d69181b65c2d5364f40845466a9e5305ca8c91b/OpenKeychain/src/main/java/org/sufficientlysecure/keychain/securitytoken/usb/UsbTransport.java#L237)), so it looks like the Nitrokey 3 cannot be supported without patching it, and I don't know whether it will be merged since the project is in maintenance only.

I was not able to get it to recognise my device on Waydroid, but this may be caused by usb issues with waydroid and not the app itself?

proninyaroslav, 2023-05-26:
@sosthene-nitrokey 
I was unable to initially connect it via USB, but I successfully used it via NFC until version 1.4.0.

szszszsz, 2023-05-29:
@proninyaroslav Since the Nitrokey 3 v1.4.0 firmware the user data are mostly moved to the external flash chip, which is not activated in the NFC mode, thus this will not work anymore unfortunately.

@sosthene-nitrokey I believe in the settings you can allow all devices. Can you check again? 
I could not make Waydroid work either, even with other USB devices (some gamepads are reportedly somehow supported?).

ciropom, 2023-05-30:
> 

> believe in the settings you can allow all devices. Can you check again?

Which settings? the OpenKeyChain (OKC) settings? I confirm that I can't add my nitrokey to OKG neither by plugging it in the phone's usb-c
I enabled "Settings -> experimental -> Allow untested usb devices" but nothing changes.
If I plug the nitrokey, the led doesn't blink and if I tap "USE TOKEN" it is not recognized.

ciropom, 2023-06-01:
I don't quite understand what NFC will be useful for, if no user data is available in NFC mode..

xundeenergie, 2023-06-06:
> Not by the current design. It was decided against it, to instead invest in the secure element-based storage security. The NFC chip gives not enough power for the additional storage and secure element use.
> 
> Perhaps in the distant future some user-selected data would be available in lower-security storage at user's choice, giving back the ability of NFC use.

Hmmm... this brakes my workflows again. I bought the nk3 to use it with openkeychain on my smartphone... and now it does not work via NFC neither via USB-C... 

This is NOT good.

ciropom, 2023-06-07:
I completely agree with xundeenergie

shaohme, 2023-06-09:
And I agree with both of them :-D

danielkrajnik, 2023-07-17:
Just to confirm I tried now nitrokey 3a NFC and 3a mini over Usb C adapter and both said:

````
This Security Key is not yet supported by OpenKeychain
```

````
Firmware version:   v1.5.0
````

@sosthene-nitrokey What do you think?

danielkrajnik, 2023-07-18:
Thank you very much for the fast response, explaining the status and link to the pull request. Glad to see this work on USB 
support. I will keep fingers crossed that NFC support will arrive at some point in the future (maybe Nitrokey 4).

daringer, 2023-07-18:
> We currently do not have plans to support OpenPGP over NFC due to power limitations.

Small addition --- I would like to emphasize that "currently" this is not the plan. We see and hear that OpenPGPCard via NFC is something that people want, we simply did not expect this. Generally this is possible on the Nitrokey 3 hardware, but this will come with trade-offs in terms of functionality and added complexity on development side. Especially there will be some foundation work needed to make this possible. Feel free to :+1: this comment, if this is an important functionality for you...

ciropom, 2023-07-19:
I think that if this comes through NFC, it would be best, but also via usb-c is OK! I buyed this token for this reason mainly. 
I was thinking to import my gpg sub-keys  in the openkeychain app through usb-c.
 
By now, I'm taking advantage of the "nk3 secrets" subcommand of nitropy that I find very useful too for TOTP and passwords.

szszszsz, 2023-07-19:
@ciropom 
By the way, would you use the `nk3 secrets` equivalent on a mobile too?

ciropom, 2023-07-19:
For sure.
It will be very cool to have a way to access secrets from android (either usb-c or NFC)

xundeenergie, 2023-08-30:
> > We currently do not have plans to support OpenPGP over NFC due to power limitations.
> 
> Small addition --- I would like to emphasize that "currently" this is not the plan. We see and hear that OpenPGPCard via NFC is something that people want, we simply did not expect this. Generally this is possible on the Nitrokey 3 hardware, but this will come with trade-offs in terms of functionality and added complexity on development side. Especially there will be some foundation work needed to make this possible. Feel free to 👍 this comment, if this is an important functionality for you...

I bought my nitrokey to use it for gpg/openkeychain to have my password-store secured on my phone with an extra hardware-token (i use pass, which encrypts passwords with gpg-keys) and i wanted to secure my ssh-connections using a hardware-token for the ssh-keys... 

both is not working. both was promised, when i orderd the key... especially for exakt THIS... 

Just having a FIDO-Key, there are cheaper tokens...

danielkrajnik, 2023-08-30:
> feedback on those changes
I don't suppose that there is any release on F-Droid to test this? I've been watching this [feature progress](https://github.com/open-keychain/open-keychain/pull/2842) for the past month, but I was hoping that it may be made available via some package.

sjlongland, 2023-08-30:
> > feedback on those changes
> 
> @sjlongland I don't suppose that there is any release on F-Droid to test this?

Sadly no, I think the pull request would need to be merged first and a beta release cut by the upstream project before we saw anything in F-Droid.  I don't have any control over this.

danielkrajnik, 2023-09-01:
No worries, fingers crossed then that it will be merged soon.

ciropom, 2024-04-03:
Something changed since my last comment.
I upgraded to firmware 1.6.0, and now openkeychain recognizes my key (if I insert it in the mini-usb slot) but I get this error now
[image]

daringer, 2024-04-12:
yes, the most recent open-keychain release should now work with the nk3... the output I see should be good news, e.g., that open-keychain can use the token, I do not know exactly if this is enough .... are you able to use the key on your nk3 with open-keychain and k-9 or something ?

ciropom, 2024-04-12:
The key is somehow recognized in openkeychain, but then fails to do everything except "search on keyservers" and doesn't view my GPG keys, even if they are there 

```
Reader ...........: 20A0:42B2:X:0
Application ID ...: D276000124010304000F9B79DABF0000
Application type .: OpenPGP
Version ..........: 3.4
Manufacturer .....: unknown
Serial number ....: 9B79DABF
Name of cardholder: Danilo Tomasoni
Language prefs ...: it
Salutation .......: 
URL of public key : keyserver.ubuntu.com
Login data .......: [not set]
Signature PIN ....: forced
Key attributes ...: ed25519 cv25519 ed25519
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 0 3
Signature counter : 1
KDF setting ......: off
Signature key ....: ..................................
      created ....: 2024-04-02 21:19:13
Encryption key....: ..................................
      created ....: 2024-04-02 21:21:44
Authentication key: ..................................
      created ....: 2024-04-02 21:22:14
General key info..: sub  ed25519/.................................. 2024-04-02 Danilo Tomasoni (unitn) <danilo.tomasoni@studenti.unitn.it>
sec#  rsa2048/..................................  created: 2010-11-16  expires: never     
ssb>  ed25519/..................................  created: 2024-04-02  expires: never     
                                card-no: 000F 9B79DABF
ssb>  cv25519/..................................  created: 2024-04-02  expires: never     
                                card-no: 000F 9B79DABF
ssb>  ed25519/..................................  created: 2024-04-02  expires: never     
                                card-no: 000F 9B79DABF
ssb#  ed25519/.................................. created: 2023-08-29  expires: never     
ssb#  cv25519/..................................  created: 2023-08-29  expires: never     
ssb#  ed25519/..................................  created: 2023-08-29  expires: never     
ssb#  rsa2048/..................................  created: 2023-05-23  expires: never     
ssb#  rsa2048/..................................  created: 2023-05-23  expires: never     
ssb#  rsa2048/..................................  created: 2023-05-23  expires: never     
ssb#  rsa2048/..................................  created: 2010-12-23  expires: never     
```

pjhfggij, 2024-04-14:
> OpenPGP is not supported over NFC. Does this happen with FIDO or is it limited to OpenPGP?

@sosthene-nitrokey are there plans to add support for GPG over NFC for "Nitrokey 4"?

ciropom, 2024-04-14:
I understand it isn't working through NFC, but openkeychain is not supporting it neither by inserting the nitrokey 3 in the phone usb-c I'm using e /os/ with microg as google play services replacement, don't know if this is related..

pjhfggij, 2024-04-14:
@ciropom hmm have you tried importing public gpg key as file?

ciropom, 2024-04-14:
no I thought it will be searching for the public key on the keyservers.. I will try

ciropom, 2024-04-14:
oh gosh that's was the key.. the public key was not loaded from keyservers. Once I imported it from file, then suddenly it was loaded.. I will try k9-mail and other stuff to confirm now..

ciropom, 2024-04-14:
i confirm that it works both with openkeychain and with k9-mail through openkeychain

pjhfggij, 2024-04-14:
Yep, I've realized that same thing today as well (after having put OKC on hold for half a year). Better late than never I guess.
@sosthene-nitrokey if only there was some NFC to top it up