Nitrokey/nitrokey-3-firmware #275: Flash nitrokey3 on LPCXpresso55S69

tarun14110, 2023-05-27:
I am planning to utilize the LPCXpresso55S69 board for development purposes and I am currently attempting to flash the Nitrokey3. However, I encountered an issue while following the instruction `lpc55 ls` command, no output was displayed. I can see the NXP board listed in the dmesg output. Based on this observation, I suspect the board might not be in bootloader mode. I followed the steps outlined in the following link (https://os.mbed.com/teams/NXP/wiki/Updating-LPCXpresso-firmware), but unfortunately, it did not resolve the problem. I would greatly appreciate any suggestions or alternative instructions to properly set up the LPCXpresso55S69 board for development and successfully flash the Nitrokey3.

tarun14110, 2023-05-29:
Yes! I am following these instructions. 
After I connect the board, I see the following through `lsusb` command `Bus 002 Device 003: ID 1fc9:0090 NXP Semiconductors LPC-LINK2 CMSIS-DAP v 5.361`. According to the documentation above, this is not what I am supposed to see. Also, I also do not see anything on running `lpc55 ls`.

Based on the product id I see in the above documentation, I assume it is because I am using the built-in LPC-LINK2 debug probe in LPCXpresso55S69 board instead of an external MCU-LINK debug probe. Any suggestions on how can I get it to work?

szszszsz, 2023-05-29:
I think you need to move one of the switches, like JP2. See the mentioned doc for the details.

tarun14110, 2023-05-29:
Thank @szszszsz! I installed the J-link firmware based on the instructions in the documentation. I then run command `make -C utils/lpc55-builder jlink` on terminal 1 and `make -C utils/lpc55-builder run` on terminal 2. I successfully was able to enter in `gdb` mode.

If I want to flash the nitrokey3 firmware without entering into debug mode and use the LPCXpresso55S69 board as nitrokey3, how can I do that? Is there a way to use J-link for that?
Thanks for helping out with this. I am just getting started with hardware, so let me know if I am missing something obvious.

robin-nitrokey, 2023-05-30:
The easiest way to run the Nitrokey 3 firmware on a development device is using a Nitrokey 3 Hacker device which you can order at shop@nitrokey.com.  We can give you some pointers to get it running on the LPCXpresso55 board, but you will have to make some changes yourself and it is not an officially supported configuration.

It is very likely that the Xpresso board has a different pinout than the actual Nitrokey hardware so this might cause some issues.  For debugging, I recommend the following steps:
- While `gdb` is running (i. e. `make run`), press `Ctrl+C` to see the code that the firmware is currently executing.  It might be stuck in the panic handler because something went wrong during the setup.
- Enable logging by setting the `FEATURES=log-all,log-semihosting` environment variable.  The log messages should show up in the `gdb` output.

Previously, we had support for the Xpresso board in the firmware, but we dropped it because we were not actually using it and had no way to test it:
- [`lpcxpresso55/button.rs`](https://github.com/Nitrokey/nitrokey-3-firmware/blob/e6b0fd0391f98917f0e654955ea5dea850027a56/runners/lpc55/board/src/lpcxpresso55/button.rs)
- [`lpcxpresso55/led.rs`](https://github.com/Nitrokey/nitrokey-3-firmware/blob/e6b0fd0391f98917f0e654955ea5dea850027a56/runners/lpc55/board/src/lpcxpresso55/led.rs)

Since then, we also added checks of the SE050 and the external flash at boot.  These could also fail on your board.

robin-nitrokey, 2024-04-24:
Closing as there is no action required at the moment.  Please open a new issue if there are specific problems.

closed 2024-04-24 by robin-nitrokey