Nitrokey/nitrokey-3-firmware #281: How to use HMAC-SHA1 Challenge Response with NK3

orolhawion, 2023-06-05:
Hi,

I just updated my NK3 to 1.5.0 and could not find any instructions on how to use HMAC-SHA1 Challenge Response? I thought the secrets app was the app to use for it, but I could not find anything in the docs. How do you guys do it?

szszszsz, 2023-06-05:
Hi!

This new feature is aimed to support KeepassXC and other compatible applications, so no CLI was added to pynitrokey for it. To set it up a new pynitrokey release is needed, which should be published this week. The specific code is here: https://github.com/Nitrokey/pynitrokey/pull/393.

Setup for HMAC slot 2 will look like this:
```
$ nitropy nk3 secrets add-challenge-response 2 (echo 1234567890123456789 |  base32)
```

Do you have any specific use case for this Yubikey's protocol? Just a reminder, that it is already possible to use HMAC challenge response through FIDO2, though I think this might be broken in the current nitropy 0.4.37:

```
$ nitropy fido2 challenge-response  --help
```

orolhawion, 2023-06-05:
I basically use it to unlock `KeepassXC` and to not type a password when I use `sudo`.

I didn't know about challenge-response through FIDO2, thanks for the hint.

AdmerStroh, 2023-06-05:
Hi,
I compiled keepassxc with https://github.com/keepassxreboot/keepassxc/pull/9397 an pynitrokey with https://github.com/Nitrokey/pynitrokey/pull/393 
Than I added a secret to the second HMAC slot with` nitropy nk3 secrets add-challenge-response 2 $(tr -dc '0-9' </dev/urandom | head -c 20 | base32)`
The secret is shown with `nitropy nk3 secrets list` but its not detected in keepassxc.

AdmerStroh, 2023-06-05:
Hey @szszszsz ,
for me, pcscd was not started on boot. Starting it manually fixed my problem and the Nitrokey 3 gets detected in keepassxc.

Thank you for that hint in the right direction! :)

szszszsz, 2023-06-05:
Thank you for testing!

szszszsz, 2023-06-06:
For the completeness I have added to my original reply a link to the low-level tests in case anyone would like to use the response in his custom solutions. The solution is generic enough to support virtually infinite Yubikey's-like HMAC slots.
However the recommended way is to use FIDO2's HMAC challenge-response, as a more standardized and flexible tool.

Closing as done. Transforming this into actionable task.

closed 2023-06-06 by szszszsz

orolhawion, 2023-06-06:
I would just like to confirm that challenge response through fido2 is broken in nitropy 0.4.37 as assumed by @szszszsz in an earlier post.

orolhawion, 2023-06-07:
With `nitropy 0.4.38` I can use the secrets app to create an HMAC-SHA1 challenge response credential.  Why is the secret limited to 20 bytes? I use a secret with 40 bytes with my yubikeys and would like to do so with NK3.

The `FIDO2` way is still broken.

szszszsz, 2023-08-19:
Just for the final note, the KeepassXC support for Nitrokey 3 was just released.

@orolhawion I've registered your request for longer secrets at https://github.com/Nitrokey/trussed-secrets-app/issues/89

bhmarscheck, 2023-12-07:
The release notes for 1.6.0 states that it "Remove challenge response authentication method". Does this mean I can't use my nk3 for unlocking my KeepassXC if I update my firmware?

robin-nitrokey, 2023-12-07:
No, these are two different things:
- The challenge-response authentication that is “removed” in 1.6.0 was used for authentication towards the device, i. e. unlocking data that is protected with a PIN.  It used a challenge-response mechanism instead of sending the user PIN to the device.  Actually, this feature was already disabled in previous releases, this release just completely removes the code that was used for it.  So the changelog is somewhat misleading here.  I will remove the line.
- The HMAC challenge-response used for KeepassXC is a special credential type and is not affected by this change.