Nitrokey/nitrokey-3-firmware #284: Document HMAC-SHA1 Challenge Response use with NK3

szszszsz, 2023-06-06:
Make sure HMAC-SHA1 Challenge Response use with NK3 is properly documented.
Recommend FIDO2 HMAC challenge-response use where possible.

Details: https://github.com/Nitrokey/nitrokey-3-firmware/issues/281

merryswiftoctopus, 2023-12-25:
Hi, I have been able to set up my NitroKey 3A for my Keepass vault; However, I do not quite understand: Is there a way to protect this secret with a PIN? Because now the key just needs to be plugged in, no further authentication required.

ChristianTackeGSI, 2023-12-26:
I never tried it. (I am actually hoping for some fido2-with-pin thing becoming an alternative to the normal password.)

I guess, the idea is that you use it as a second factor in addition to the knowledge of the main keepass password. So you still need to know something (enter password into your keepass) and own something (insert the nitrokey).

daringer, 2024-01-02:
currently this is not possible and @ChristianTackeGSI explains the overall concepts precisely... nevertheless it's still a valid request, which is tracked here: https://github.com/Nitrokey/pynitrokey/issues/487