Nitrokey/nitrokey-3-firmware #285: x230 with Heads reported to not work with NK3

szszszsz, 2023-06-07:
x230 with Heads is reported to not work with NK3:

> I recently purchased Nitrokey 3 NFC and tried to use it with my X230, but when I tried to change HOTP, the system wrote that the Librem Key device was not found, it did not seem to recognize the N3K. Problem in NK3 or in Heads?

Source: https://github.com/osresearch/heads/issues/1354#issuecomment-1581344094

cc @daringer

szszszsz, 2023-07-06:
Interim note: I got reports, that everything is fine with the recent Heads builds.

daringer, 2023-07-07:
release including nk3 support: https://github.com/Nitrokey/heads/releases/tag/v2.1
upstream PR: https://github.com/osresearch/heads/pull/1417

closing as done

closed 2023-07-07 by daringer