Nitrokey/nitrokey-3-firmware #286: Luks with systemd-cryptenroll panics after a reboot

sosthene-nitrokey, 2023-06-08:
See https://github.com/Nitrokey/nitrokey-3-firmware/issues/283#issuecomment-1581517261

> Environment:
> 
> 1. Arch Linux
> 2. latest version of available packages: gpg 2.2.41, kwallet 5.106.0, kwalletmanager 23.04.1 at this very moment
> 3. full disk encryption with unlocking using FIDO token, the token was enrolled using `systemd-cryptenroll`
> 4. KDE wallet encrypted using GPG keys
> 5. GPG keys reside on Nitrokey 3A NFC
> 6. FW version is 1.5.0, but the same was observed with 1.4.0 as well

> if I unlock the wallet in KDE successfully and then reboot the machine without replugging the token, the token panics with red LED on disk FIDO decryption.

Unlocking the KDE wallet requires GPG, there seems to be some conflicts between GPG and FIDO-authenticator

pfactum, 2023-06-09:
I can also confirm this issue with token panic on a freshly installed system on a different machine.

sosthene-nitrokey, 2023-06-27:
Like #283 it was fixed by https://github.com/Nitrokey/nitrokey-3-firmware/pull/289

closed 2023-06-27 by sosthene-nitrokey